thonny/thonny #646: Simple open file to edit from command line failing

mMerlin, 2019-01-03:
Fedora 27 : 4.18.19-100.fc27.x86
zsh 5.4.1
Python 3.6.6
virtualenv 16.0.0
Thonny 3.0.8
Tk 8.6.8

thonny is installed `pip install thonny` inside the active virtual environment

Specifying a relative file name path on the command line when starting thonny is failing with an assertion error `assert os.path.isabs(name)`, on the first line of `normpath_with_actual_case(name: str)` in common.py.  I traced through the call stack with `python -m pdb pdb_thonny.py test.py`, and verified that `test.py` is passed unmodified all the way down.  Obviously, not an absolute path name, so it dumps.  Simplified trace:

```ts
Traceback (most recent call last):
  File "…/thonny/workbench.py", line 164, in __init__
    self._editor_notebook.load_startup_files()
  File "…/thonny/code.py", line 484, in load_startup_files
    self.show_file(filename)
  File "…/thonny/code.py", line 672, in show_file
    editor = self.get_editor(filename, True)
  File "…/thonny/code.py", line 710, in get_editor
    return self._open_file(filename)
  File "…/thonny/code.py", line 698, in _open_file
    editor = Editor(self, filename)
  File "…/thonny/code.py", line 56, in __init__
    self._load_file(filename)
  File "…/thonny/code.py", line 161, in _load_file
    filename = normpath_with_actual_case(filename)
  File "…/thonny/common.py", line 207, in normpath_with_actual_case
    assert os.path.isabs(name)
AssertionError
```

Either the file name needs to be expanded to an absolute path before calling normpath_with_actual_case, or that function needs to handle simple file names.

Opening the file once thonny is running works fine.

mMerlin, 2019-01-04:
Sure, but **where**?  How soon in that nested set of calls should it be done?  I don't know which pieces of code might be reused for something else, and could break if the expand was done in the wrong place.

I suspect this will also mix with where thonny believes / sets the current working directory.  Which is not necessarily where it was actually started from.  At least according to some tests reporting os.getcwd().  A quick look around seems to imply that the `working_directory` entry in `.thonny/configuration.ini` is used, so would need to expand to absolute before that takes affect.  And that is probably why it is insisting at the bottom of that stack, that the file name must have an absolute path.

aivarannamaa, 2019-01-04:
The `working_directory` thing is related to magic commands (%Run and %cd) sent to the backend. I believe front-end process never changes (or cares about) its cwd, so it should be safe to expand relative paths anywhere in the front-end code.

I would recommend doing it in the beginning of `thonny.code.EditorNotebook.get_editor`.

closed 2019-01-05 by mMerlin