thonny/thonny #648: alternate load path for asttokens failing on attribute error

mMerlin, 2019-01-05:
Fedora 27 : 4.18.19-100.fc27.x86
Python 3.6.6 (64-bit)
Thonny 3.0.8 16a5ec1d168a 
Tk 8.6.8

`python3 -m thonny` in the repository clone folder, shows traceback information in the opened thonny shell window.

```ts
Traceback (most recent call last):
  File "…/thonny/thonny/backend_launcher.py", line 70, in <module>
    VM().mainloop()
  File "…/thonny/thonny/backend.py", line 118, in __init__
    self._load_shared_modules(init_msg["frontend_sys_path"])
  File "…/thonny/thonny/backend.py", line 343, in _load_shared_modules
    load_module_from_alternative_path(name, frontend_sys_path)
  File "…/thonny/thonny/backend.py", line 2564, in load_module_from_alternative_path
    spec.loader.exec_module(module)
AttributeError: 'zipimport.zipimporter' object has no attribute 'exec_module'
Backend terminated (returncode: 1)
Use 'Stop/Restart' to restart the backend ...
```

Checking older commits, this error first showed up recently in

```tx
commit abd64abec03eaee44597242742fc65da12f30d51
Author: Aivar Annamaa <aivar.annamaa@gmail.com>
Date:   Sun Dec 30 13:08:54 2018 +0200

    Share frontend's asttokens with backend
```

Running that version gets almost the same traceback.  Just the last 2 line numbers are off by -2.  A bit of debugging, to load those modules one at a time, instead of in the loop, finds that "asttokens" is the culprit.  Minor modification on `abd64abe` gets

```ts
Traceback (most recent call last):
  File "…/thonny/thonny/backend_launcher.py", line 70, in <module>
    VM().mainloop()
  File "…/thonny/thonny/backend.py", line 118, in __init__
    self._load_shared_modules(init_msg["frontend_sys_path"])
  File "…/thonny/thonny/backend.py", line 344, in _load_shared_modules
    load_module_from_alternative_path("asttokens", frontend_sys_path)
  File "…/thonny/thonny/backend.py", line 2567, in load_module_from_alternative_path
    spec.loader.exec_module(module)
AttributeError: 'zipimport.zipimporter' object has no attribute 'exec_module'
Backend terminated (returncode: 1)
Use 'Stop/Restart' to restart the backend ...
```
The only reference I can find with grep to zipimport is in the misc_analyzers.py plugin.

aivarannamaa, 2019-01-09:
Thank you for the report! 

I'll try to test and fix Thonny in Fedora before the end of the month.

aivarannamaa, 2019-01-24:
Did you `pip install asttokens` or `dnf install python3-asttokens` ?

mMerlin, 2019-01-24:
Not specifically, but it is installed.  I assume it was installed when I installed thonny (before starting to work with the repo)

```sh
% pip show asttokens
Name: asttokens
Version: 1.1.13
Summary: Annotate AST trees with source code positions
Home-page: https://github.com/gristlabs/asttokens
Author: Dmitry Sagalovskiy, Grist Labs
Author-email: dmitry@getgrist.com
License: Apache 2.0
Location: /home/phil/.local/lib/python3.6/site-packages/asttokens-1.1.13-py3.6.egg
Requires: six
Required-by: thonny

% sudo dnf history userinstalled | grep "python"
python3-lxml-4.1.1-1.fc27.x86_64
python3-qrcode-5.1-7.fc27.noarch
python3-tkinter-3.6.6-1.fc27.x86_64
python3-virtualenv-16.0.0-4.fc27.noarch
```

There is more than I like installed in the main python environment.  In general, I prefer to run things from virtual environments to keep the package scope limited, but things accumulate over time, especially for *tools*.
```sh
% pip list  
Package           Version    
----------------- -----------
asn1crypto        0.23.0     
astroid           1.5.2      
asttokens         1.1.13     
Beaker            1.5.4      
beautifulsoup4    4.6.0      
blivet            2.1.11     
blivet-gui        2.1.7      
Brlapi            0.6.6      
cffi              1.10.0     
chardet           3.0.4      
coverage          4.4.2      
cryptography      2.3        
cssselect         0.9.2      
cupshelpers       1.0        
cycler            0.10.0     
decorator         4.0.11     
docutils          0.14       
fros              1.1        
gpg               1.10.0     
html5lib          0.999999999
httpie            0.9.4      
humanize          0.5.1      
hunspell          0.5.4      
idna              2.5        
iniparse          0.4        
IPy               0.81       
isc               2.0        
isort             4.2.15     
jedi              0.13.2     
langtable         0.0.38     
lazy-object-proxy 1.3.1      
lxml              4.1.1      
Mako              1.0.6.dev0 
MarkupSafe        0.23       
matplotlib        2.0.0      
mccabe            0.6.1      
mypy              0.650      
mypy-extensions   0.4.1      
ntplib            0.3.3      
numpy             1.13.3     
olefile           0.45.1     
ordered-set       2.0.0      
parso             0.3.1      
pid               2.1.1      
Pillow            4.3.0      
pip               18.1       
ply               3.9        
pwquality         1.4.0      
py                1.4.34     
pycairo           1.15.3     
pycodestyle       2.4.0      
pycparser         2.14       
pycups            1.9.72     
pycurl            7.43.0     
pydbus            0.6.0      
pyenchant         1.6.10     
Pygments          2.2.0      
pygobject         3.26.1     
pyinotify         0.9.6      
PyIscsi           1.0        
pykickstart       2.41       
pylint            1.7.5      
pyparsing         2.1.10     
pyparted          3.11.0     
pyperclip         1.7.0      
pyserial          3.4        
PySocks           1.6.7      
pytest            3.2.3      
pytest-runner     3.0        
python-augeas     0.5.0      
python-dateutil   2.6.0      
python-dmidecode  3.12.2     
python-meh        0.43       
pytz              2017.2     
pyudev            0.21.0     
pyxdg             0.25       
qrcode            5.1        
requests          2.20.0     
requests-file     1.4        
requests-ftp      0.3.1      
rpm               4.14.2.1   
scudcloud         1.65       
sepolicy          1.1        
setools           4.1.1      
setroubleshoot    1.1        
setuptools        37.0.0     
simpleline        0.6        
six               1.11.0     
slip              0.6.4      
slip.dbus         0.6.4      
sos               3.5.1      
SSSDConfig        1.16.3     
systemd-python    234        
thonny            3.0.8      
typed-ast         1.1.1      
urllib3           1.24.1     
virtualenv        16.0.0     
webencodings      0.5.1      
wrapt             1.10.10  
```

aivarannamaa, 2019-01-24:
I can reproduce the problem when I install asttokens with easy_install (instead of pip).

I think I got it fixed now. Please reopen if this doesn't work for you!

closed 2019-01-24 by aivarannamaa

mMerlin, 2019-01-24:
syncing my fork, and retry worked.  No error messages on startup.