thonny/thonny #651: Using PyGameZero

PolarBear456, 2019-01-08:
Hi there. I hope this is where I should put any issues I find, as I'm a little bit of a noob. So I would like to program the MagPi Pygame Zero series in Thonny, as it is an excellent Python IDE. I'm probably using the newest version of Pygame Zero, Thonny and everything else. **I've already posted a thread [here](https://www.raspberrypi.org/forums/viewtopic.php?f=32&t=230611), but I don't seem to get some progress.** Please take a look at that thread to understand what I'm saying. So will it work in Thonny, do I have to update anything or is it just a fact that I can't use Thonny for Pygame Zero?
Thank You

aivarannamaa, 2019-01-09:
Hi!

Indeed, Thonny 3.0.8 has a bug that doesn't allow `import pgzrun`. In the next version this bug will be fixed and furthermore -- there will be special "PyGame Zero mode", which allows skipping `import pgzrun` and `pgzrun.go()` altogether.

acbart, 2019-01-21:
I hadn't heard of PygameZero prior to this, but it looks awesome! In this new Pygame Zero mode, it sounds like it'll be easy to integrate into a classroom very early in the course. How will PyGame Zero mode be activated?

aivarannamaa, 2019-01-22:
There is a toggle-item in the Run menu for this.

In the simple UI mode the menu is not visible, but I intend to enhance the implementation such that in this mode Thonny deduces from the code whether it should be run in normal mode or in PGZero mode.

closed 2019-01-23 by aivarannamaa