thonny/thonny #652: the thonny wiki is still on bitbucket [solution]

mMerlin, 2019-01-08:
Both bitbucket and github wikis are themself git repositories.  I do not see a way to fork those, or setup a pull request, but the following can be used to copy the bitbucket wiki content, and all of the change history, over to the github wiki.  This will wipe out any existing change history from the github wiki, but that is still minimal.

A warning to go with this.  The actual document content is copied directly.  However, the image links in the pages still point to the original images on bitbucket.  Those will still need to be copied over to github at some point, and the links updated.  I have not thought of a way to automate that.  The links themselves are easy to locate.  In the wiki repository created for this move process, `grep "bitbucket" *` nicely shows them all.  I count 25.

```sh
git clone https://bitbucket.org/plas/thonny.git/wiki
cd wiki
git remote set-url origin https://github.com/thonny/thonny.wiki.git
git rm __not_used4.rst
«any initial edits»
git add -A
git commit -m "prepare to move to github wiki"
git push --force
```

I was doing some other playing in the wiki in my thonny repository fork @ https://github.com/mMerlin/thonny/wiki, so it is not an accurate represenation of the results.  I had another repo though that the wiki was not yet in use for, so https://github.com/mMerlin/templates/wiki does currently show what the result should look like.  That can stay for awhile, but I will be wiping it out at some point.  As noted in the above script, I deleted the `__not_used4.rst` used page, and did a few minor tweeks to DeploymentOptions page.  Just enough to prove my changes were getting posted along with everything else.

Reviewing the wiki page content, some of it seems to be well out of date.  [SeparateInstall](https://bitbucket.org/plas/thonny/wiki/SeparateInstall) says that Python 3.5 or later is needed, however, [Windows](https://bitbucket.org/plas/thonny/wiki/Windows), has a note about having to use python 3.4.4 on windows XP.  I assume, if that still works, that those instructions will also install an older version of thonny.  Whatever the result, a little more detail in that area would be good.  Probably a good idea to extract that whole block to its own `Windows XP` page, and just link to it from the `Windows` page.

I saw a comment in the CONTRIBUTING document that the main developers are not native English speakers.  I see some signs of that in the wiki.  Not incorrect content, but clumsy wording.  Not a lot, and I have seem content from people that ARE native English speakers that is a lot worse.

mMerlin, 2019-01-09:
The advantage of copying this way, is the the full history of changes is maintained in the [wiki] repository.  You can use standard git tools to see what changed and when.

Since it **is** a git repo, you can also keep the full history **AND** replace everything (or as much as you can handle) at once.  In my `initial edits` step, just change as much as you want.  That can be updated pages, new pages, deleted pages.  Whatever you want.  I should test this, but I think you can load images, and link to them that way too.  Pulling the old images out of bitbucket, to maintain the history, is probably the biggest problem.

aivarannamaa, 2019-01-12:
Done! Thanks for the suggestion!

aivarannamaa, 2019-01-12:
Can you please make an edit to a wiki page so that I can check whether I get a notification?

closed 2019-01-12 by aivarannamaa

mMerlin, 2019-01-12:
Minor tweaks done and committed for SeparateInstall page.  You should have a notification for that.  Looks like you renamed the page files from `.rst` to `.rest`.  I am not seeing the history information from before the move.  The information is all still there, but not visible in the page history.  That stops with the new file got created.  git commands, with the right options, will see the full history.

aivarannamaa, 2019-01-22:
I did not receive the notification from the Github, but I asked https://blogtrottr.com/ to notify me whenever https://github.com/thonny/thonny/wiki.atom changes (see https://stackoverflow.com/questions/8407917/how-can-you-track-or-be-notified-of-changes-to-github-wikis for more).

Thank you for your help!