thonny/thonny #658: issue with mypy v0.660

timsoftgit, 2019-01-17:
mypy v0.650 works fine, but if you install mypy v0.660 then on completion of running a simple "hello world" program, you get the message (on the shell that called thonny)

> WARNING: MyPy: /example/file.py: error: INTERNAL ERROR -- please report a bug at https://github,com/python/mypy/issues version: 0.660
> /example/file.py: : note: please use --show-traceback to print a traceback when reporting a bug

the program is just 1 line
`print("hello world")`

aivarannamaa, 2019-01-22:
Thank you for the report! I'll investigate it later.

aivarannamaa, 2019-01-28:
I'm not able to reproduce this problem in Ubuntu 18.04 with MyPy 0.660.

aivarannamaa, 2019-01-28:
I forgot to add I was using Python 3.6.7 in a venv.

timsoftgit, 2019-01-31:
I have just compared the source tgz of mypy0.650 and mpy0.660, and v0.660 is missing the typeshed directory and contents. This may explain the difference. Is it needed?

timsoftgit, 2019-01-31:
I've found that downloading the source tar.gz from git misses the typeshed files with mypy>0.650
i got aversion with the files from 
https://files.pythonhosted.org/packages/a4/f6/032e015285905243cff94625367c7a20ba0acbae8e802b740c8a876b2f39/mypy-0.660.tar.gz
using this one removes the errors I was getting. I've raised a bug on mypy git to highlight this issue incase there is a fix at the github end, but we can close it here, and I've submitted the 3.1.0 thonny to slackbuilds.org so that should become live  after the weekend.

closed 2019-01-31 by timsoftgit