thonny/thonny #659: Automatic horizontal scrollbar is missing in Ubuntu

aivarannamaa, 2019-01-17:


aivarannamaa, 2019-01-18:
It is visible only when editor is loaded with too long lines.

It is disappearing when making editor wider but it doesn't reappear when making narrow.

aivarannamaa, 2019-01-26:
Actually it also appears when you type the line so that it becomes bigger. It does not appear when you grow the line from the beginning.

Same behavior in Linux and Windows.

lurch, 2020-05-31:
I was finding this _really_ annoying, so decided to dig into it, and found the problem! However it still needs some more thought before it's properly fixed :confused: 
Here's what I've found so far...

The horizontal scrollbar isn't appearing because it got broken by https://github.com/thonny/thonny/commit/f763d022b472aaa9f588e12abfea0e94217638c6
```
        elif float(lo) > 0.001 or float(hi) < 0.009:
            # with >0 and <1 it occasionally made scrollbar wobble back and forth
```
I believe there was probably a small logic slip-up made here and the second number should actually have been 0.999 (i.e. `1.0 - 0.001`).

Indeed if I change https://github.com/thonny/thonny/blob/master/thonny/ui_utils.py#L859 to
```
        elif float(first) > 0.001 or float(last) < 0.99:
```
then the horizontal scrollbar appears as expected when a long text-line is visible in the window. Woohoo!!! :tada: 

The reason that the current version of Thonny only shows the horizontal scrollbar when you scroll "off the right-side of the window" (e.g. by using the cursor keys on a long line, or typing extra characters at the end of a line), is because this then scrolls the editor-view to the right, which makes `first` > 0.001 (and so the `elif` check passes). And the scrollbar doesn't disappear again until `float(first) <= 0.0 and float(last) >= 1.0`.

However with the fix in place I also discovered why this scrollbar "occasionally wobbles" (and I reproduced the "# TODO: this can make GUI hang or max out CPU when scrollbar wobbles back and forth" comment in the commit mentioned above).

The way the AutoScrollbar works (with my fix) is that if there are any lines in the editor's current view of the file that are longer than the editor's width, the scrollbar gets displayed (because `last` is now < 1.0). But if all the lines in the editor's current view of the file are _shorter_ than the editor's width (even if there are lines "off-screen" that are longer), the scrollbar gets hidden (because `last` now equals 1.0).

For a simplified example, imagine we have a file looking like:
```
1:345
2:345
3:345
4:345
5:345
6:3456789
7:345
```
and imagine that Thonny's editor is currently showing a width of 6 characters and a height of 5 lines.
* If line 1 is the first line visible in the editor (i.e. only lines 1-5 are visible), then all the (visible) lines are shorter than 6 characters, so `last` equals 1.0, and the horizontal scrollbar isn't shown.
* If Thonny is then scrolled down so that lines 3-7 are visible, then line 6 is longer than 6 characters, so `last` is now less than `1.0` and so with my fix above the horizontal scrollbar is now shown.
* The problem comes when lines 2-6 are visible in the window. Line 6 is longer than 6 characters, so `last` is now less than `1.0` and the horizontal scrollbar is shown. With _just the right amount_ of vertical scrolling, the horizontal scrollbar appearing might then cause line 6 to be obscured by the scrollbar, and so not visible in the editor. With line 6 not being visible in the editor, all the visible lines are < 6 characters, making `last` now 1.0, so the scrollbar gets removed. With the scrollbar now removed, line 6 is visible again, so last is now < 1.0, so the scrollbar gets added. With the scrollbar added, line 6 is not visible again.....  Ooops, infinite loop :wink:  (i.e. the scrollbar isn't "wobbling" it's actually getting repeatedly added and removed.)

I don't have time to dig into this further, but hopefully what I've found so far is useful. Maybe the "fix" would be to include any lines obscured by the horizontal scrollbar when calculating `last`? :shrug:

aivarannamaa, 2020-06-01:
Thank you for this investigation! I hope to find time for Thonny in couple of days.

aivarannamaa, 2020-06-05:
> I believe there was probably a small logic slip-up made here and the second number should actually have been 0.999

Indeed, that was the problem.

Thank you for explaining how to reproduce the wobbling. I'd prefer not to calculate scrollbar ranges myself, so I think it should be solved such that the hide/show procedure is triggered only from a user action (eg. text change, resizing the parent component, vertical scroll)

lurch, 2020-06-05:
> Indeed, that was the problem.

:tada: 

> I think it should be solved such that the hide/show procedure is triggered only from a user action (eg. key up and mouse up)

Would that mean that if the user clicks on the vertical scrollbar and holds down the left-mouse-button to drag it up or down, then the horizontal scrollbar would only appear (if necessary) when the user releases the mouse button? That's probably only a minor inconvenience if it fixes the wobbling problem though :+1:

jharris1993, 2022-02-13:
Ref:
https://github.com/thonny/thonny/issues/659#issuecomment-636466768

Lurch has done it again!

I noticed that the issue was opened on Jan 18, 2019 and Lurch's fix was posted May 31, 2020, it has not yet made it into the current code as of this instant date, (Feb. 13, 2022 checked against version 3.3.14)

I think everyone agrees that this is an extremely frustrating problem.&nbsp; I am curios, is there a timeline for this to be included in the code?

Does someone have to create a pull-request to trigger this?
(If that's the case, I'll gladly do it.)

Do you need a patch?&nbsp; Here it is!
```
--- ./Old/ui_utils.py	2022-02-13 22:52:23.146508754 +0300
+++ ./New/ui_utils.py	2022-02-13 22:45:41.490957915 +0300
@@ -941,7 +941,7 @@
     def set(self, first, last):
         if float(first) <= 0.0 and float(last) >= 1.0:
             self.grid_remove()
-        elif float(first) > 0.001 or float(last) < 0.009:
+        elif float(first) > 0.001 or float(last) < 0.999:
             # with >0 and <1 it occasionally made scrollbar wobble back and forth
             self.grid()
         ttk.Scrollbar.set(self, first, last)
```

Let me know if there is anything I can do to help get this going.

Thanks!

closed 2022-02-15 by aivarannamaa