thonny/thonny #665: PEP8 support

crja72, 2019-01-24:
Please, add PEP8 suggestions / autoformating

Saijin-Naib, 2019-01-24:
Yeah, PEP8 and pylinter are two things I really have come to enjoy in VSCode. I'd like to see them in Thonny as well.

bendavis78, 2019-01-25:
If this is added, it should be optional, disabled by default. I'm a PEP8 perfectionist myself, but when it comes to teaching beginners, I find it adds too much cognitive overhead and distracts from the real learning objectives.

Saijin-Naib, 2019-01-26:
Totally fair.

However, as a self-taught utter noob, I find the PEP8/linting helpful. I have no idea what good practices are really since I have no one to interact with when I'm coding, so I use it as a last-pass after I've struggled some code together to make it more as it should be.

acbart, 2019-02-07:
I was playing around with making a plugin that provided a toolbar with a few buttons for running pytest, coverage.py, flakes, and autopep8. My vision was that these would pop up an HTML report in the user's web browser. The on-demand button model seems a little less intrusive to me than constant linting/CI for that level. Does this seem suitable for your uses?

aivarannamaa, 2019-02-07:
Sounds nice! Can I see the code and try it out?

acbart, 2019-02-07:
I'm afraid it's not very close to being done. I mostly just got a new menu bar to appear and have a button that can run coverage on the current file. I was gauging whether there was interest in continuing it.

aivarannamaa, 2019-02-07:
Do you think new bar is required? Don't you want to include the commands in the main toolbar?

acbart, 2019-02-07:
I figured I'd break something if I tried that - having them in the main
toolbar would be more ideal. Although I could also see the argument against
clogging up that bar.

aivarannamaa, 2019-02-07:
`get_workbench().add_command(..., include_in_toolbar=True)` should do the trick.

> I could also see the argument against clogging up that bar.

I assume you intend to provide this feature as separately installable plug-in. In this case I don't see problems with adding more icons to the toolbar.

Simple access to running tests might be useful even in the main package, but I'd like to avoid pytest dependency. One option would be adding `%Test` magic command to the main package (together with menu item and toolbar button), which uses pytest if available and a simple custom test-runner otherwise. If it detects more complex test cases (with parameterization etc.), it would suggest installing pytest. What do you think?

How do you intend to determine the scope of testing when user clicks the button? Always running the tests in the current script only?

acbart, 2019-02-21:
Having a basic fallback sounds like a good idea to me!

Yeah, the assumption would be to run the current file and that tests would be in it or inside of an imported file. No magic to go searching for separate tests.

bendavis78, 2019-12-13:
The best argument for adding linting in Thonny is that it helps students identify errors and correct them before running their code.

A great example of this is learning variable scope. If you type a variable name that's not available in scope, you get immediate feedback. That immediate feedback helps the student learn about scope more quickly.

aivarannamaa, 2019-12-14:
Thonny does use Pylint (View => Assistant). The feedback appears when user executeds the code, because I believe distractions during editing are not beneficial for beginners.

aivarannamaa, 2020-06-02:
Current closest workaround for executing PEP 8 checks:

- Install `pycodestyle` (the recommended successor of `pep8`) for Thonny GUI process (eg. from Tools => Manage plug-ins)
- enter `!pycodestyle my_script.py` in Thonny's shell

michiel0001, 2022-03-11:
I am a beginner and using Thonny. Support of a Style Guide would help me a lot if it would only show the most important errors. You decide what the most important style elements are 🧐. If it would also show the style guide item explanation it would contribute to the learning. :"How does it work? Can I see a good and a bad example?"

abawer, 2022-10-11:
It would be very nice if Thonny could support autopep8 run over files within a click of a (maybe customized) menu button. Something like allowing to run `autopep8 -i %file` over the files in the projects or current viewed file

rootsmusic, 2023-11-07:
Try the [thonny-edulint](https://github.com/GiraffeReversed/thonny-edulint#readme) plugin.