thonny/thonny #677: Debugging fails before starting

juanfal, 2019-01-31:
It's funny, the program runs perfectly well without debugging but debug doesn't work with it.  It sends you these messages:

    >>> %cd '/Users/juanfc/Dropbox/Automata Celular/AE-2'
    >>> %Debug automata_evolutivo.py -n -f ACMnegativo -g 10
    PROBLEM WITH THONNY'S BACK-END:

    Traceback (most recent call last):
      File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/thonny/backend.py", line 1208, in execute_source
        root = self._prepare_ast(source, filename, mode)
      File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/thonny/backend.py", line 1531, in _prepare_ast
        self._tag_nodes(root)
      File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/thonny/backend.py", line 2111, in _tag_nodes
        last_child = ast_utils.get_last_child(node)
      File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/thonny/ast_utils.py", line 126, in get_last_child
        and hasattr(node.slice, "step")
    AssertionError
    >>>

aivarannamaa, 2019-01-31:
Can you send me the program?

Which Thonny version are you using? Mac bundle or installed with pip?

aivarannamaa, 2019-02-01:
I could reproduce the problem with following program:

```
x[:, :, :]
```
I'll try to fix it this weekend.

juanfal, 2019-02-01:
:)