thonny/thonny #678: Extend syntax coloring to f-strings

phr0gger, 2019-02-07:
Coloring the entire f-string as string is misleading when it contains expressions.
Atom does color expressions in f-strings, so does VIM I believe.

aivarannamaa, 2019-02-07:
Agreed.

Can you look up how Atom or VIM does it? If there is a regex for finding these expressions, I could implement it sooner.

phr0gger, 2019-02-12:
There might be other ways, but the MagicStack folks are pushing [their MagicPython package]( https://marketplace.visualstudio.com/items?itemName=magicstack.MagicPython) for Atom, Sublime, and VSCode.

vilhok, 2021-09-22:
What's the status on this one?

aivarannamaa, 2021-09-22:
> What's the status on this one?

Still in the TODO list.

TomEpsilon, 2022-01-15:
Ah, what about `fr'(\d{{1,2}})/(\d{{1,2}})/(\d{{4}}|\d{{2}})'`?