thonny/thonny #679: Raspberri Pi

nikospaz, 2019-02-07:
i get this error

Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/thonny/__init__.py", line 65, in launch
    bench = workbench.Workbench(server_socket)
  File "/usr/lib/python3/dist-packages/thonny/workbench.py", line 150, in __init__
    self.reload_themes()
  File "/usr/lib/python3/dist-packages/thonny/workbench.py", line 1108, in reload_themes
    self._apply_ui_theme(preferred_theme)
  File "/usr/lib/python3/dist-packages/thonny/workbench.py", line 1049, in _apply_ui_theme
    self._register_ui_theme_as_tk_theme(name)
  File "/usr/lib/python3/dist-packages/thonny/workbench.py", line 1038, in _register_ui_theme_as_tk_theme
    settings = settings()
  File "/usr/lib/python3/dist-packages/thonnycontrib/pi/__init__.py", line 189, in pix
    bgr = re.search (r"selected_bg_color:#([0-9a-fA-F]*)", line, re.M).group(1)  # @UndefinedVariable
AttributeError: 'NoneType' object has no attribute 'group'

aivarannamaa, 2019-02-07:
Which Thonny's version are you using?

aivarannamaa, 2019-02-08:
Please select "About" from the "Help" menu to be sure about the version!

spl237, 2019-02-08:
I've submitted a fix for this to our apt repo - should be available over the weekend sometime.

nikospaz, 2019-02-08:
how i fix it? it doesn't start to see about from help

spl237, 2019-02-08:
Run Appearance Settings (under Preferences in the main Raspbian menu). Go to the System tab, and set a highlight and text colour using the boxes on that page. That should then enable Thonny to start.

Alternatively, wait a few days and then do a "sudo apt-get update" followed by "sudo apt-get dist-upgrade" from the command line to update Thonny with the fix.

closed 2019-09-01 by aivarannamaa