thonny/thonny #680: Debugger: PROBLEM WITH THONNY'S BACK-END:

vinniec, 2019-02-09:
I use a code which always works if normally started, but not during debugging.
The strange things is that at first the debugger was working, but after a while it stopped working and it hasn't worked anymore.

I use thonny 3.0.8 (thonny-3.0.8-x86_64.tar.gz) for linux (ubuntu 17.10),  decompressed into a directory and started directly from ./bin.
I tried to delete and unpack thonny again, I tried to delete the directory of additional libraries (~/.local/lib/python3.7/) I tried to delete thonny settings (~./.config/Thonny) but nothing.
So I tried to isolate the piece of code that's causing problems and it's really weird:

`("kMGTPE" if True else "KMGTPE")[1]`
Without round brackets (which are not necessary) thonny does not give any error during debugging! And I repeat, at the beginning debugging also worked but after a while it stopped working.

The content of the shell is this:
`>>> %Debug prova.py
PROBLEM WITH THONNY'S BACK-END:

Traceback (most recent call last):
  File "/home/vinnie/software/thonny/thonny/lib/python3.7/site-packages/thonny/backend.py", line 1136, in execute_source
    root = self._prepare_ast(source, filename, mode)
  File "/home/vinnie/software/thonny/thonny/lib/python3.7/site-packages/thonny/backend.py", line 1467, in _prepare_ast
    self._tag_nodes(root)
  File "/home/vinnie/software/thonny/thonny/lib/python3.7/site-packages/thonny/backend.py", line 2036, in _tag_nodes
    last_child = ast_utils.get_last_child(node)
  File "/home/vinnie/software/thonny/thonny/lib/python3.7/site-packages/thonny/ast_utils.py", line 155, in get_last_child
    and hasattr(node.slice, "step")
AssertionError
>>> 
`

aivarannamaa, 2019-02-09:
Please check whether it works in version 3.1.0: https://github.com/thonny/thonny/releases/tag/v3.1.0

vinniec, 2019-02-09:
Ok, the problem looks like it's disappeared. Thank you very much!

Another stupid thing that doesn't create any problems and maybe isn't even a bug, thonny writes this in the shell when it meets the exit() instruction:
`
Traceback (most recent call last):
  File "/home/vinnie/software/thonny/thonny/lib/python3.7/site-packages/thonny/backend.py", line 1569, in _trace
    return self._trace_and_catch(frame, event, arg)
  File "/home/vinnie/software/thonny/thonny/lib/python3.7/site-packages/thonny/backend.py", line 1635, in _trace_and_catch
    self._handle_progress_event(frame, pseudo_event, {}, last_custom_frame.node)
  File "/home/vinnie/software/thonny/thonny/lib/python3.7/site-packages/thonny/backend.py", line 1657, in _handle_progress_event
    self._respond_to_commands()
  File "/home/vinnie/software/thonny/thonny/lib/python3.7/site-packages/thonny/backend.py", line 1781, in _respond_to_commands
    self._current_command = self._fetch_next_debugger_command()
  File "/home/vinnie/software/thonny/thonny/lib/python3.7/site-packages/thonny/backend.py", line 1335, in _fetch_next_debugger_command
    cmd = self._vm._fetch_command()
  File "/home/vinnie/software/thonny/thonny/lib/python3.7/site-packages/thonny/backend.py", line 882, in _fetch_command
    line = self._original_stdin.readline()
ValueError: I/O operation on closed file.
`

aivarannamaa, 2019-02-09:
I guess it is harmless, but it is not very nice. Can you create a separate issue for this?

(BTW. I'll make another bugfix release tonight: 3.1.1)

closed 2019-02-09 by aivarannamaa

vinniec, 2019-02-09:
Ok