thonny/thonny #681: Problem con exit() instruction

vinniec, 2019-02-09:
Problem encountered with thonny 3.1.0 (and also 3.0.8) with ubuntu 17.10.
Thonny writes this in the shell when it meets the exit() instruction (although this does not cause any problems with the execution of the code.)

Traceback (most recent call last):
File "/home/vinnie/software/thonny/thonny/lib/python3.7/site-packages/thonny/backend.py", line 1569, in _trace
return self._trace_and_catch(frame, event, arg)
File "/home/vinnie/software/thonny/thonny/lib/python3.7/site-packages/thonny/backend.py", line 1635, in _trace_and_catch
self._handle_progress_event(frame, pseudo_event, {}, last_custom_frame.node)
File "/home/vinnie/software/thonny/thonny/lib/python3.7/site-packages/thonny/backend.py", line 1657, in _handle_progress_event
self._respond_to_commands()
File "/home/vinnie/software/thonny/thonny/lib/python3.7/site-packages/thonny/backend.py", line 1781, in _respond_to_commands
self._current_command = self._fetch_next_debugger_command()
File "/home/vinnie/software/thonny/thonny/lib/python3.7/site-packages/thonny/backend.py", line 1335, in _fetch_next_debugger_command
cmd = self._vm._fetch_command()
File "/home/vinnie/software/thonny/thonny/lib/python3.7/site-packages/thonny/backend.py", line 882, in _fetch_command
line = self._original_stdin.readline()
ValueError: I/O operation on closed file.

chpurdy, 2019-03-24:
Is this still an issue in 3.1.2?  I can't seem to duplicate it on mine.  Maybe a code snippet of an example program that causes the error?

aivarannamaa, 2019-09-01:
Not reproducible anymore

closed 2019-09-01 by aivarannamaa