thonny/thonny #682: Extend printing to shell window

ebonak, 2019-02-09:
The printing of source code was a welcome addition.

Would it be possible to extend printing to the shell window as well? This way it would be easy to capture and print longer program runs without having to resort to several (possibly disjointed) screen shots. 

My students would be able to submit their program output very easily by printing the scrollable contents of the shell window.

aivarannamaa, 2019-02-09:
This would require another print command, because probably some users want to print the scipt only. And there is the question of how much of the shell output to print.

What about requiring your students to copy relevant part of the output as multiline string literal to the end of the script before printing the script?

ebonak, 2019-02-09:
I just envisioned the same print command would act on whatever window was currently active, i.e., if my cursor was in the edit window, it would print that, if it was in shell window, it would print it.

As for how much to print, I'd print from the last clear screen/^L or perhaps whatever the user selected (I think the former would be easier to implement and just fine).

I'll have to think about your suggestion re the multi-line string literal and maybe give it a try. It would be relatively easy if they were doing a straight type of report, but for instance they currently have an interactive menu driven program with some sub-menus and error conditions etc, I'm not quite how/if that would work or if that would mess up/modify the formatting of the output which is also an important factor.  I'll have to give it a test run.