thonny/thonny #683: Unable to have Function Keys always show on Touchbar Macbook Pro's

chris-larsen, 2019-02-09:
When configuring Function keys to always show in MacOS Mojave 10.14.2 (system preferences -> keyboard -> shortcuts -> function keys -> add "thonny" app), function keys still don't show when Thonny is active. 

[image]

chris-larsen, 2019-02-13:
Anyone able to reproduce this? Not having the issue with PyCharm, VSCode or any other apps really.

aivarannamaa, 2019-02-13:
What does "Function keys to always show" mean? Is is something about newer Mac models?

I'm not a Mac user myself, but I have an old MacBook Pro 2010 for testing Thonny. How should I reproduce this?

Does the same problem occur when using IDLE?

chris-larsen, 2019-02-13:
2016 and newer Macbook models have a LCD touch strip instead of physical F-keys. This touchstrip is contextual based on what app your in and different app provides different virtual buttons for things. If your Macbook is equipped with this, dubbed the Touchbar by Apple, you have to hold FN for your F-keys to show as they're now virtual button elements drawn onto the touchbar. Lots of developers prefer having F-keys always show, and Apple allows this in the screenshot I posted above. 

This feature works fine in IDLE.

aivarannamaa, 2019-02-14:
Thanks for the explanation! I'll check how they have solved it in IDLE (it uses same UI framework).

aivarannamaa, 2021-02-27:
Please try whether https://github.com/thonny/thonny/releases/download/v3.3.5/thonny-3.3.5_alt.pkg works better for you. It's with an experimental Universal 2 variant of Python 3.9.2 with Tk 8.6.11.

aivarannamaa, 2022-08-26:
Assuming this is fixed in 4.0.0. Please re-open if it is not.

closed 2022-08-26 by aivarannamaa