thonny/thonny #684: v3.1.1 doesn't work with Windows

nathanRamaNoodles, 2019-02-11:
I am getting an error when I load Thonny:
[image]
'Nonetype' attribute has no attribute 'call'

aivarannamaa, 2019-02-11:
Similar issue was discussed here: #639
Do same suggestions work for you?

Gefionious, 2019-03-20:
I had the same issue and the fix was the same (#639). I think the CSR is installed as part of a pretty popular bluetooth adapter driver or something, I have not completely traced the source but that was frustrating. There was no mention on the FAQ or the Wiki. Also I downloaded Thonny 3.1.2 so problem still there.

Thank you. My son loves using your tool and suggested I start using it, but it was very frustrating not being able to launch. I ended up uninstalling all my other Python versions, libraries and IDE's thinking it was some sort of conflict somewhere, obviously in the right general direction but all wasted effort. Sigh.

aivarannamaa, 2019-03-20:
Thank you for the feedback! I'll make sure the fix will be in version 3.2

aivarannamaa, 2019-06-07:
Duplicate of #639

closed 2019-06-07 by aivarannamaa