thonny/thonny #685: Print cuts lines

juanfal, 2019-02-11:
Long lines are cut (not wrapped) when printed.  It's source failure, but perhaps could be admitted wrapping better than cutting out parts of the code as a more helpful hint for badly formatted source code with long lines.

aivarannamaa, 2019-02-11:
I think I found a solution. Can you please check if it works in your browser as well?

After you issue the Print command, close the print dialog and save the html file. Open the file in a text editor and add following to the CSS block of the `pre` element (after `counter-reset: line;`):

```
    white-space: pre-wrap;       /* Since CSS 2.1 */
    white-space: -moz-pre-wrap;  /* Mozilla, since 1999 */
    white-space: -pre-wrap;      /* Opera 4-6 */
    white-space: -o-pre-wrap;    /* Opera 7 */
    word-wrap: break-word;       /* Internet Explorer 5.5+ */
```

Save and open the html in browser. Does it wrap lines now?

juanfal, 2019-02-11:
Yes it does the wrapping, but it is doing something strange with the with of the numbers column, wrapping the numbers too:

aivarannamaa, 2019-02-11:
Does it wrap only line numbers over 99? I think I have solution for this as well: https://github.com/thonny/thonny/issues/687

commit 950f26a3 (2019-02-12): FIXED #685: Print cuts lines
closed 2019-02-12 by aivarannamaa