thonny/thonny #686: Debug at 100% CPU lasting very long

juanfal, 2019-02-11:
On OSX, last 3.1.1, debug takes very very long time when entering jumping over not so long imports.  It seems even stuck

aivarannamaa, 2019-02-11:
Can you provide an example program?

juanfal, 2019-02-11:
Well, in fact it is the very same one I privately sent to you.  It happens when

	import AE, AE_FILE

-

aivarannamaa, 2019-02-11:
Thanks! I can confirm that this problem occurs also on Linux. I'll investigate it later.
Meanwhile I recommend to use faster debugger (Shift+F5).

juanfal, 2019-02-11:
No words.  Thanks!

-

closed 2019-02-12 by aivarannamaa

aivarannamaa, 2019-02-13:
Please check whether it works in 3.1.2 for you: https://github.com/thonny/thonny/releases/download/v3.1.2/thonny-3.1.2.dmg

juanfal, 2019-02-13:
It now works like a charm!!!!!

The listings now don’t wrap the line numbers and the ⌘-⌥-E and ⌘-⌥-S work nices

Wow!

Thanks Aivar.
If you come to Málaga/Spain consider it done you have free room+beers :)

- Juan F.

commit dcf9f898 (2019-02-12): FIXED #686: Stepping over user modules can be too slow