thonny/thonny #690: Python 'is' operator issue when interepreting script

yolandamay, 2019-02-13:
[image]

As seen in the above screenshot, Thonny has an issue when computing is in script but has no problems when doing the same operation in command line.

aivarannamaa, 2019-02-13:
Did you expect that the comparison is always `True`? I'm not familiar with Python's rules about interning strings, but I wouldn't make any assumptions in this situation. 

Maybe you can find an answer here: http://guilload.com/python-string-interning/

yolandamay, 2019-02-13:
Is the string interned when it's run from the window? Is that why it is returning `True` when it should have returned `False` like it does in the interpreter?

aivarannamaa, 2019-02-13:
No problem. It's surprising behavior indeed -- I would have expected it returns same result in both cases. Anyway, Thonny has no say about this -- it delegates all work to standard Python interpreter.