thonny/thonny #695: Thonny's window title bar does not change to dark mode in macOS Mojave

phr0gger, 2019-02-17:
This is Thonny 3.1.2:
[image]

aivarannamaa, 2019-02-17:
Do you have IDLE installed? Does it use correct titlebar?

phr0gger, 2019-02-17:
No, IDLE also keeps the window title bar bright. Ist it possibly a GTK issue?

aivarannamaa, 2019-02-17:
Not GTK, but Tk (the toolkit used by both Thonny and IDLE via Tkinter).

I'm afraid we need to wait until official Python interpreter (used in Thonny) gets built with current Tk version (8.6.9.1)

phr0gger, 2019-02-17:
oops, I was confused. Good to know. Thank you!

aivarannamaa, 2021-02-27:
Fixed by upgrading to Tk 8.6.11. 

Experimental build available in https://github.com/thonny/thonny/releases/download/v3.3.5/thonny-3.3.5_alt.pkg (bundled with Universal 2 binaries of Python 3.9.2)

closed 2021-02-27 by aivarannamaa