thonny/thonny #696: Possible bug: crash in turtle graphics: invalid slave specification

jdeisenberg, 2019-02-19:
Here is the traceback received by a person running Thonny on Darwin 16.7.0, Tk8.6.8, Python 3.7.1 (64 bit) on a Macintosh when running the attached file (converted to txt to make github happy)
[khishen_lina_clock.py.txt](https://github.com/thonny/thonny/files/2878032/khishen_lina_clock.py.txt) This occurs when the program finishes, but it appears that the graphics window is still on-screen in the background.
```
Traceback (most recent call last):
  File "/Users/linakhishen/Desktop/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/thonny/workbench.py", line 1432, in event_generate
    handler(event)
  File "/Users/linakhishen/Desktop/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/thonny/plugins/debugger.py", line 1051, in _handle_toplevel_response
    _current_debugger.close()
  File "/Users/linakhishen/Desktop/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/thonny/plugins/debugger.py", line 165, in close
    self._last_frame_visualizer.close()
  File "/Users/linakhishen/Desktop/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/thonny/plugins/debugger.py", line 529, in close
    self._decorate_editor_title("")
  File "/Users/linakhishen/Desktop/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/thonny/plugins/debugger.py", line 521, in _decorate_editor_title
    self.editor, self.editor.get_title() + suffix
  File "/Users/linakhishen/Desktop/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/thonny/code.py", line 695, in update_editor_title
    self.tab(editor, text=title)
  File "/Users/linakhishen/Desktop/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.7/lib/python3.7/tkinter/ttk.py", line 903, in tab
    return _val_or_dict(self.tk, kw, self._w, "tab", tab_id)
  File "/Users/linakhishen/Desktop/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.7/lib/python3.7/tkinter/ttk.py", line 299, in _val_or_dict
    res = tk.call(*(args + options))
_tkinter.TclError: Invalid slave specification .!frame.!automaticpanedwindow.!automaticpanedwindow2.!editornotebook.!editor4
```

aivarannamaa, 2019-02-19:
Thank you for the report!

Was it a single case or did it happen several times? Only with this turtle program?

jdeisenberg, 2019-02-19:
The program I attached crashes every time. I am not sure about other programs; I will have to contact the student who wrote this and ask them to run their other programs.

aivarannamaa, 2019-07-07:
Can't reproduce anymore. Probably got fixed by b9ced71e92d7ab57cd3e9aab7d513b0e83305473

closed 2019-07-07 by aivarannamaa