thonny/thonny #697: Create portable Windows variant of Thonny

aivarannamaa, 2019-02-19:


Xalaca, 2019-02-25:
Congratulations for this amazing software which I consider the perfect one for learning Python.

I've just followed your instructions on how to create a portable version of Thonny and ended up realizing that unfortunately it is not totally portable because of the fact that the packages manager uses a fixed file.

When I got here to submit the idea for this issue I got this wonderful surprise that you have set it as a milestone.

Thank you a lot and I'm sure that this feature will help spread Thonny and the python language, or I at least will

aivarannamaa, 2019-04-18:
Partly done in eee3db4

aivarannamaa, 2019-06-20:
Installing packages to USB drive will be dealt with in version 4.0

aivarannamaa, 2019-06-20:
Related to #718

aivarannamaa, 2019-06-21:
Documented here: https://github.com/thonny/thonny/wiki/DeploymentOptions#portable-version

closed 2019-06-21 by aivarannamaa

mexterng, 2024-06-02:
I want to change some features and add new ones that go deeper into the system, beyond what plugins can do.

How can I create the same directory and file structure from my changed source files as in the portable release for Windows? The file https://github.com/thonny/thonny/blob/master/packaging/windows/create_installer.bat uses Thonnyrunner. How do I update these necassary files?