thonny/thonny #698: Shell output font appearance

Ilis, 2019-02-21:
Shell output font is not as defined in Options.

It's better to be as chosen font for Editor, or configurable.

[image]

willbprog127, 2019-03-02:
Yes, I was just looking for this in the options.  Please allow this to be customized. :+1:

sazun, 2019-03-16:
Yes, please. I am using unicode geometric characters for printing trees and the actual font is not suitable.

aroberge, 2019-05-28:
I have made a mock-up (UI only) of what it could look like, as an addition to the current Shell option tab. Would this be acceptable - assuming I made the required code changes to actually make it work!

[image]

aivarannamaa, 2019-05-29:
Looks good, but what do you think of putting same items on the "Theme & Font" page? Family and size boxes just below editor's family and size boxes and Shell preview below Editor preview?

To me this would feel more natural, especially considering that Syntax theme should affect the shell preview as well.

> I just realize that as I used Black to format the modules I worked on, it did change the formatting quite a bit. If needed, I could try to undo these formatting changes and only add the new code

It would be easier to spot the changes if you did this. On the other hand, I think I need to switch the whole project to Black to avoid similar hassles in the future (#735).

If you agree about changing the location of these boxes, then it's probably cleanest if I reject this PR, format all the code with Black and then you'll reapply your changes on top of the blackened master. 
What do you think?

aroberge, 2019-05-29:
I certainly have no problem with redoing the work (it should be much faster now) and put this on the "Theme & Font" page.   I thought that the page "Theme & Font" would have been the right one to use ... but thought it was already full enough.

In my current PR, I had to switch the preview to using a single tk.Label.  When I tried to use either a CodeView or a ShellText (or something similar), some exception would be generated whenever I dismissed the options menu and tried to interact with the shell.    I will try to see I can use three Labels for the stdout/stdin/stderr and pick up their colours from the theme.

aivarannamaa, 2019-05-29:
>  When I tried to use either a CodeView or a ShellText (or something similar), some exception would be generated ...

You can use a plain tk.Text for now. I can check later how to use ShellText instead. Probably I've made some implicit assumptions about it being a singleton.

aroberge, 2019-05-29:
> When I tried to use either a CodeView or a ShellText (or something
> similar), some exception would be generated ...
>
> You can use a plain tk.Text for now. I can check later how to use
> ShellText instead. Probably I've made some implicit assumptions about it
> being a singleton.
Very good, I will.

aivarannamaa, 2019-05-29:
I'll do the blackening right away

aivarannamaa, 2019-05-29:
Code is now blackened with 100 char line length (it is also specified in the pyproject.toml file, so simple `black thonny` should work)

aroberge, 2019-05-29:
Many themes (Tomorrow Night Eighties,
Default Dark Blue,
Default Dark Green,
Tomorrow,
Tomorrow Night,
Tomorrow Night Blue,
Tomorrow Night Bright,
Tomorrow Night Eighties )
do not include values for stdin/stderr/stdout.

When trying to get themes, I get a callable which, if called, returns a dict for the themes. This works fine.
However, there is one exception: Tomorrow Night Eighties returns a dict instead of a callable.

With some try/except blocks, I can work around these problems and got this issue "fixed", by using the values from the Default Dark theme ... which work reasonably well for all themes **except** Tomorrow.

I could create a pull request with this work around - but it might be better to set the missing values in the various themes first so that I would not have to include any try/except block which are really a kludge.

aivarannamaa, 2019-05-29:
Syntax themes (and also UI themes) use inheritance, so the final set of options will have those values. It's possible to query current syntax theme with `thonny.codeview.get_syntax_options_for_tag`. But it might be better to create a `thonny.codeview.SyntaxText` and insert demo text with suitable tags.

(I still believe that it's also possible to use ShellText, but it may need some refactoring).

aroberge, 2019-05-29:
Ok, I will try with `thonny.codeview.get_syntax_options_for_tag`.  I might be able to do that today and create a new pull request.

aivarannamaa, 2019-05-30:
Fixed by @aroberge. Thank you!

closed 2019-05-30 by aivarannamaa