thonny/thonny #700: All files isn't all files

Farcrada, 2019-02-28:
When you go to `File` -> `Open...`  and select: `all files (*.*)`, it is in fact not all files unless it's got a `.` in the name, and thus has an extension of some sort. However opening it with Thonny via the context menu works just fine.

I like to create my programs with a shebang and no extension simply because I think it looks nice in the terminal. Having said that; `all files` isn't then recognizing all files, but only those with a dot somewhere in the name.

aivarannamaa, 2019-02-28:
Thanks for the report! Which platform are you using?

I remember Mac having problems with `*`-patterns (but it may have changed with newer Tk versions)

Farcrada, 2019-02-28:
Linux Mint 19.1 Installed via Software Manager

commit e2e27f0e (2019-03-01): FIXED #700: All files filter in file dialogs don't show all files in Linux
closed 2019-03-01 by aivarannamaa