thonny/thonny #702: Use 'Stop/Restart' to restart the backend ...

DrCai, 2019-03-01:
[image]

DrCai, 2019-03-01:
I know why, because I added PYTHONPATH to my environment variable.

aivarannamaa, 2019-03-01:
Might be...
Hopefully fixing #639 will help in your case as well.

Ilis, 2019-03-07:
Using Stop sign for restart is a bit weird. The actual result of "Stop/Restart" action is Restart, so it's better be some sort of Restart icon, not Stop.

aivarannamaa, 2019-03-09:
> Using Stop sign for restart is a bit weird. The actual result of "Stop/Restart" action is Restart, so it's better be some sort of Restart icon, not Stop.

The main reason for putting this button to the toolbar was to provide quick means for getting out of infinite loops. Just restarting is only required when Thonny does someting bad (or when switching MicroPython devices). Therefore, I would say in most important cases the main action is Stop and Restart is a bonus.

closed 2019-04-10 by aivarannamaa

victorkolis, 2020-11-24:
Whenever I run the pygame module and quit the window I've created this message pops up:

**Backend terminated or disconnected. Use 'Stop/Restart' to restart.**

aivarannamaa, 2020-11-25:
@victorkolis, can you provide a minimal program that demonstrates this problem?

Which OS, Thonny version and Pygame version are you using?