thonny/thonny #706: Thonny hangs on close sometimes

Ilis, 2019-03-10:
Thonny hangs now and then when closing (using Close button in window, Exit menu or exit() in command line). Window freezes until one terminate one of two python-3 processes using System monitor.

Linux Mint 18.3 and 19.1, Thonny 3.1.2

aivarannamaa, 2019-03-12:
I haven't met such behavior myself. Please update this thread if you notice some regularities about this!

abrothman, 2019-06-05:
I am experiencing the same issue. It happens only after trying to _run_ some code, either with the run button/menu, or by typing into the console.

Raspberry Pi, Raspbian, Thonny 3.1.0

aivarannamaa, 2019-06-06:
I recently noticed that in Linux Thonny may hang on close when it has copied something to the clipboard. Can you please check this by copying some text from the editor and shell and closing the window.

abrothman, 2019-06-06:
Confirmed! Copying to the clipboard causes a hang on exit.

aivarannamaa, 2019-07-06:
Initial workaround: 4ea86d82c3d855b30fd26811ea549684e2a7488f

closed 2019-07-07 by aivarannamaa