thonny/thonny #708: Confusing error pop-up when indexing incorrectly with Pandas (missing debug help file?)

gvwilson, 2019-03-17:
I have installed Pandas for use with Thonny.  I run the following code:

```
import pandas as pd
data = pd.read_csv('somefile.csv')
data[0] # note: should not work
```

Thonny displays a pop-up error dialog with the following:

```
Traceback (most recent call last):
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/thonny/workbench.py", line 1449, in event_generate
    handler(event)
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/thonny/assistance.py", line 138, in handle_toplevel_response
    self._explain_exception(msg["user_exception"])
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/thonny/assistance.py", line 178, in _explain_exception
    + _error_helper_classes["*"]
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/thonny/assistance.py", line 176, in <listcomp>
    for helper_class in (
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/thonny/assistance.py", line 491, in __init__
    super().__init__(error_info)
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/thonny/assistance.py", line 478, in __init__
    self.last_frame_module_source = read_source(self.last_frame.filename)
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/thonny/common.py", line 252, in read_source
    with tokenize.open(filename) as fp:
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.7/lib/python3.7/tokenize.py", line 447, in open
    buffer = _builtin_open(filename, 'rb')
FileNotFoundError: [Errno 2] No such file or directory: 'pandas/_libs/hashtable_class_helper.pxi'
```

I interpret this to mean that it does not have or cannot find some sort of auxiliary debugging help file for Pandas. When the dialog is closed, the Shell pane in Thonny displays this:

```
Traceback (most recent call last):
  File "/Users/gvwilson/Library/Python/3.7/lib/python/site-packages/pandas/core/indexes/base.py", line 2657, in get_loc
    return self._engine.get_loc(key)
  File "pandas/_libs/index.pyx", line 108, in pandas._libs.index.IndexEngine.get_loc
  File "pandas/_libs/index.pyx", line 132, in pandas._libs.index.IndexEngine.get_loc
  File "pandas/_libs/hashtable_class_helper.pxi", line 1601, in pandas._libs.hashtable.PyObjectHashTable.get_item
  File "pandas/_libs/hashtable_class_helper.pxi", line 1608, in pandas._libs.hashtable.PyObjectHashTable.get_item
KeyError: 0

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<pyshell>", line 1, in <module>
  File "/Users/gvwilson/Library/Python/3.7/lib/python/site-packages/pandas/core/frame.py", line 2927, in __getitem__
    indexer = self.columns.get_loc(key)
  File "/Users/gvwilson/Library/Python/3.7/lib/python/site-packages/pandas/core/indexes/base.py", line 2659, in get_loc
    return self._engine.get_loc(self._maybe_cast_indexer(key))
  File "pandas/_libs/index.pyx", line 108, in pandas._libs.index.IndexEngine.get_loc
  File "pandas/_libs/index.pyx", line 132, in pandas._libs.index.IndexEngine.get_loc
  File "pandas/_libs/hashtable_class_helper.pxi", line 1601, in pandas._libs.hashtable.PyObjectHashTable.get_item
  File "pandas/_libs/hashtable_class_helper.pxi", line 1608, in pandas._libs.hashtable.PyObjectHashTable.get_item
KeyError: 0
```

which is marginally more relevant and useful.

Is there a way as an instructor to create and provide whatever auxiliary file would make this more comprehensible?

aivarannamaa, 2019-03-17:
Thank you for the report! I'll try to reproduce it later.

gvwilson, 2019-03-17:
thanks - I'm happy to help write the auxiliary file (if indeed that's how this works)

closed 2019-03-20 by aivarannamaa

aivarannamaa, 2019-03-20:
Thonny's Assistant tries to read the source of last frame of a stack-trace, in order to provide more information for diagnosing the error. In this case the source of last frame was a [pxi file](https://github.com/cython/cython/wiki/FAQ#what-is-the-difference-between-a-pxd-and-pxi-file-when-should-either-be-used), which (obviously) could not be read as a normal py-file.

commit 3d2ebcfb (2019-03-20): FIXED #708: Pandas error causes confusing pop-up