thonny/thonny #709: Finding/changing current working directory

gvwilson, 2019-03-17:
I am unable to determine how Thonny decides when to change current working directory for the Python session, or how to re-set it: `os.getcwd()` tells me where I am, and `os.chdir()` will change the directory, but there doesn't appear to be a clear relation between what files I open in the editor pane and what my current working directory is. What is the recommended way to find/change this?

aivarannamaa, 2019-03-17:
At the moment on each run Thonny changes to the directory of the script.

(In the next version there will be an option which allows Files view to control working directory)

LGCarmonaS, 2019-08-03:
has anything changed?

aivarannamaa, 2019-08-03:
Thank you for reminding this!

In Thonny 3.2 the active directory in Files view becomes the current directory for the backend process. One can choose from the options (Run & Debug) whether running a file should change that directory automatically.

closed 2019-08-03 by aivarannamaa

Zandar1958, 2020-07-17:
I am getting started and would like to know how I can set up my working directory to point to my "Google Drive" file?  T.

aivarannamaa, 2020-07-18:
@Zandar1958 you should click on "This computer" in the header of the file browser and then double click on desired drive