thonny/thonny #710: error when initializing back end

Eltuta, 2019-03-21:
I am experiencing  problem with Thonny.

I have used it for about a month now, but as of yesterday, when I launch the application, the run button is inactive and throws an error " error when initializing back end"

Am learning python and your assistance will be highly appreciated.

Am not sure which program is causing the error but have got this from the front end log.

ERROR: Error when initializing backend
Traceback (most recent call last):
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\workbench.py", line 474, in _start_runner
    self._runner.start()
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\running.py", line 81, in start
    self._check_alloc_console()
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\running.py", line 516, in _check_alloc_console
    shell=True,
  File "C:\Program Files (x86)\Thonny\lib\subprocess.py", line 775, in __init__
    restore_signals, start_new_session)
  File "C:\Program Files (x86)\Thonny\lib\subprocess.py", line 1178, in _execute_child
    startupinfo)
FileNotFoundError: [WinError 2] The system cannot find the file specified

aivarannamaa, 2019-03-21:
Is it possible that you have accidentally deleted `python.exe` from `C:\Program Files (x86)\Thonny\`?

Do you use an antivirus program? If yes, then which one?

Eltuta, 2019-03-21:
I have kaspersky internet security which is very aggressive I must say.

Eltuta, 2019-03-21:
Although the signature are not updated as I have not renewed the license.

Eltuta, 2019-03-21:
I have checked, python exe id still available in program files of Thonny.

Eltuta, 2019-03-21:
I also have sublime text running on my PC but it also has thrown errors. Could it be a related problem

[WinError 2] The system cannot find the file specified
[shell_cmd: python -u "E:\Semester 2\LTAT.03.001 - Computer Programming\Solutions WK 2\H2-Shoesize.py"]
[dir: E:\Semester 2\LTAT.03.001 - Computer Programming\Solutions WK 2]
[path: C:\ProgramData\Anaconda3;C:\ProgramData\Anaconda3\Library\mingw-w64\bin;C:\ProgramData\Anaconda3\Library\usr\bin;C:\ProgramData\Anaconda3\Library\bin;C:\ProgramData\Anaconda3\Scripts;C:\Program Files (x86)\Python\Python37;C:\Program Files (x86)\Python\Python37\Scripts;C:\Users\Henry O'Dola\AppData\Local\Programs\Microsoft VS Code\bin]
[Finished]

Eltuta, 2019-03-31:
Hallo Aivar,
This is a follow up to the above error.
Have you been able to deal with "clear shell" ?,  I think it would be the best solution to my current problem.
Regards,
Henry

aivarannamaa, 2019-04-10:
Hi!

Unfortunately I haven't been able to reproduce this problem. I suggest you try next version (to be released in a month or two) to see if this resolves your problem.

> Have you been able to deal with "clear shell" ?,  I think it would be the best solution to my current problem.

What do you mean by `deal with "clear shell"` ?

closed 2019-09-01 by aivarannamaa

narektigranyan, 2022-03-18:
there is no configuration.ini file