thonny/thonny #712: project support?

segalion, 2019-03-22:
Hello. 
I have been seen this project and seems factastic. I love lightweigth & multiplatform apps.

But I miss some minimum project support...

As Thonny support tabed files, (that idle not) at least first begin with sessions (i.e. auto-open all files from last session could be great). When you are working with more than 5 files its a great help.

Some drag&drop features could be great (i.e files to open).

aivarannamaa, 2019-03-23:
Hi!

Have you tried Files view? ("View" menu => "Files"). In the next version there will be a change which allows you to "Go into" a directory such that the files view only displays one directory. I hope this will be useful as a lightweight representation of current project.

> begin with sessions (i.e. auto-open all files from last session could be great).

Good idea, I'll add an option for this.

segalion, 2019-03-25:
Thanks.
Im starting to use thonny, and I like it more and more. I dont know why all people says its only for beginners. In my opinion, it could be perfect for relative big projects... You are showing the world the great power of the reviled tkinter.

Please I would to make simple suggestions/ideas:
- For lightweight project support could be very useful to has "favorite folders", that 
1. be allways accesible (maybe on top of all folder trees, or in a separate frame inside files frame), and permanent between sessions.
2. get only name folder (hiding the up-tree but xpandable down-tree if exist).
3. Minimum context menu to "add to" and "remove" from favorites.

If you let me, another great things that I use heavily (I think Im a bad programer because I take a lot of time testing and debugging than coding)...
- When selecting text from code, if you make ctrl-f, copy the selected text in inputbox "find".
- A little input combobox remenber last searched words.
- ctrl-shift-f for "search in all" open documents if focus on code, or in favorite folders if focus 
- And the last... option to autocreate repeated code (rich snippets). I explain it a little:
In example, if I select in the code a variable 'myvar' , with a simple key-combination (i.e ctrl-shift-1, 2, 3...) it will insert a line just after into the code (with same indent) that write:
   for i in k:       
      myvar = myfunc(i)   
      print( "myvar=%s" % myvar))

The config could be as simple as : (with tkinter bind and jinja2 notation)

'<Control-Shift-Key-1'>: 'print("{{ selected_text }}=%s" % {{ selected_text }})'

And the last comment (sorry, maybe seems Im wrinting a book inside a github issue):
I dont understand "Focus editor" and "Focus Shell" menu command (you can get it with a simple clic on each frame), but realated to this, when I run programs that use 'pynput' and print-at-console all global keyspressed, I cant edit code anymore (while running), because it force focus on shell window on every key. Maybe its a bug.

aivarannamaa, 2019-06-07:
Reopening more files will be implemented under #722