thonny/thonny #716: Kivy plugin broken in 3.1.2

pidou46, 2019-04-02:
Hello,

I've just downloaded the last version of Thonny 3.1.2

[image]

Then I install the kivy plug-in 1.10.1

I'm able to import kivy, but the basic 'hello world' code sample (or any other) produce some error:

`[INFO   ] [Logger      ] Record log in F:\users\slestang\.kivy\logs\kivy_19-04-02_9.txt
[INFO   ] [Kivy        ] v1.10.1
[INFO   ] [Python      ] v3.7.2 (tags/v3.7.2:9a3ffc0492, Dec 23 2018, 22:20:52) [MSC v.1916 32 bit (Intel)]
[INFO   ] [Factory     ] 194 symbols loaded
[INFO   ] [Image       ] Providers: img_tex, img_dds, img_pil, img_gif (img_sdl2, img_ffpyplayer ignored)
[INFO   ] [Text        ] Provider: pil(['text_sdl2'] ignored)
[CRITICAL] [Window      ] Unable to find any valuable Window provider.
sdl2 - ImportError: DLL load failed: %1 n’est pas une application Win32 valide.
  File "F:\users\slestang\AppData\Roaming\Python\Python37\site-packages\kivy\core\__init__.py", line 59, in core_select_lib
    fromlist=[modulename], level=0)
  File "E:\Thonny\lib\site-packages\thonny\backend.py", line 305, in _custom_import
    module = self._original_import(*args, **kw)
  File "F:\users\slestang\AppData\Roaming\Python\Python37\site-packages\kivy\core\window\window_sdl2.py", line 26, in <module>
    from kivy.core.window._window_sdl2 import _WindowSDL2Storage
  File "E:\Thonny\lib\site-packages\thonny\backend.py", line 305, in _custom_import
    module = self._original_import(*args, **kw)

[CRITICAL] [App         ] Unable to get a Window, abort.

`
Is there a work around ?

Thanks

aivarannamaa, 2019-04-02:
I assume you meant kivy package, not plug-in. 
Did you follow Kivy installation instructions: https://kivy.org/doc/stable/installation/installation-windows.html ?

pidou46, 2019-04-02:
Thanks for your answer.

No, I install it from the Thonny's plug-ins manager : package Kivy

It use to work with earlier versions though

aivarannamaa, 2019-04-02:
> No, I install it from the Thonny's plug-ins manager

This is actually meant for Thonny's plug-ins, not for normal packages used by your programs. The right place is Tools => Manage packages. But if you haven't changed the default interpreter then both can be used.

Can you send me the list of packages you see in "Tools => Manage packages" window?

pidou46, 2019-04-02:
Here it is:

[image]

aivarannamaa, 2019-04-02:
According to the [docs](https://kivy.org/doc/stable/installation/installation-windows.html), you have some kivy dependencies missing.

Try installing also `pypiwin32`, `kivy.deps.sdl2` and `kivy.deps.glew`

pidou46, 2019-04-02:
[image]

Still get the same error

pidou46, 2019-04-02:
Sorry, no it's not the same error:

'[INFO   ] [Logger      ] Record log in F:\users\slestang\.kivy\logs\kivy_19-04-02_21.txt
[INFO   ] [Kivy        ] v1.10.1
[INFO   ] [Python      ] v3.7.2 (tags/v3.7.2:9a3ffc0492, Dec 23 2018, 22:20:52) [MSC v.1916 32 bit (Intel)]
[INFO   ] [Factory     ] 194 symbols loaded
[INFO   ] [Image       ] Providers: img_tex, img_dds, img_pil, img_gif (img_sdl2, img_ffpyplayer ignored)
[INFO   ] [Text        ] Provider: pil(['text_sdl2'] ignored)
[CRITICAL] [Window      ] Unable to find any valuable Window provider.
sdl2 - ImportError: DLL load failed: Le module spécifié est introuvable.
  File "F:\users\slestang\AppData\Roaming\Python\Python37\site-packages\kivy\core\__init__.py", line 59, in core_select_lib
    fromlist=[modulename], level=0)
  File "E:\Thonny\lib\site-packages\thonny\backend.py", line 305, in _custom_import
    module = self._original_import(*args, **kw)
  File "F:\users\slestang\AppData\Roaming\Python\Python37\site-packages\kivy\core\window\window_sdl2.py", line 26, in <module>
    from kivy.core.window._window_sdl2 import _WindowSDL2Storage
  File "E:\Thonny\lib\site-packages\thonny\backend.py", line 305, in _custom_import
    module = self._original_import(*args, **kw)

[CRITICAL] [App         ] Unable to get a Window, abort.

"Le module spécifié est introuvable." -> enable to find the specified module

pidou46, 2019-04-02:
It's seems to be a kivy related question, nothing to do with Thonny, so I will ask for help on the kivy side.

Thanks for your help.

Skiski, 2019-05-21:
I have a similar problem with matplotlib. I made a "custom" version of Thonny since I don't have administrator rights on the system and I use pip in the interpreter to install matplotlib and pygame.

import pip._internal as pip
packages=["pygame","matplotlib","sympy"]
pip.main(["install","--no-color"]+packages+["--upgrade"])

But when I try to use matplotlib:
>>> import matplotlib
Traceback (most recent call last):
  File "<pyshell>", line 1, in <module>
  File "C:\Users\romain.janvier\Downloads\thonny-3.1.2-portable-win\lib\site-packages\thonny\backend.py", line 305, in _custom_import
    module = self._original_import(*args, **kw)
  File "C:\Users\romain.janvier\Downloads\thonny-3.1.2-portable-win\lib\site-packages\matplotlib\__init__.py", line 200, in <module>
    _check_versions()
  File "C:\Users\romain.janvier\Downloads\thonny-3.1.2-portable-win\lib\site-packages\matplotlib\__init__.py", line 194, in _check_versions
    module = importlib.import_module(modname)
  File "C:\Users\romain.janvier\Downloads\thonny-3.1.2-portable-win\lib\importlib\__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
ImportError: DLL load failed: %1 n’est pas une application Win32 valide.

I think it is because pip installed the 64 bit version of matplotlib.
Pugame and numpy seem to work.

It worked with Thonny 3.0.8.

I'll try to update matplotlib with my old version of Thonny to see if it is because of a bug in matplotlib.

Skiski, 2019-05-21:
Ok, it's a problem with matplotlib 3.1.0.

aivarannamaa, 2019-09-01:
Can't reproduce in version 3.2
Please reopen if problem still exists

closed 2019-09-01 by aivarannamaa