thonny/thonny #717: edition extremely slow with "Hilight options"

segalion, 2019-04-02:
I have experienced that thonny (windows10 +thonny v.3.1.2) is extremely slow printing typed characters, when more highligt options are enabled. 

Specially syntax and matching names slow text a lot (with 10 files in tabs). 
If enable local variables it makes so unresponsable that make its directly unusable.

aivarannamaa, 2019-04-06:
Thanks for the report!

I could reproduce this in an earlier version but can't in 3.1.2.

Which platform are you using? Did you install Thonny as a bundle or with pip?

segalion, 2019-04-08:
Problem is on Windows10 + python 3.6.7 + thonny 3.1.2 installed with pip

I have thonny running on ubuntu18 with bundled 3.7 and it works fine.

aivarannamaa, 2019-04-08:
You could try whether upgrading `jedi` package helps. Latest Thonny bundles have jedi 0.13.2.

segalion, 2019-04-08:
I am on jedi 0.13.3. Maybe downgrade?

Requirement already up-to-date: jedi in
c:\users\t148888\appdata\local\programs\python\python36\lib\site-packages
(0.13.3)
Requirement already satisfied, skipping upgrade: parso>=0.3.0 in
c:\users\t148888\appdata\local\programs\python\python36\lib\site-packages
(from jedi) (0.3.4)

aivarannamaa, 2019-04-08:
Don't downgrade -- if the problem is with jedi 0.13.3, then I need to fix it. I'll check it out tonight.

aivarannamaa, 2019-04-08:
Actually, if you don't mind, then please do, it would still give information :)

segalion, 2019-04-08:
Downgraded
python.exe -m pip install jedi==0.13.2

Give same results: speed when typing is slooooooow when lot of higligh
options enabled.
So posible is not jedi version related.

aivarannamaa, 2019-09-01:
Covered by #877

closed 2019-09-01 by aivarannamaa