thonny/thonny #720: Wrong place for the "Program arguments" menu entry

juanfal, 2019-04-04:
The "Program arguments" functionality, a really convenient and very well implemented one, is, however, situated under the "View" menu.  This is unexpected as one thinks of Running when consider the arguments to the program execution.  Shouldn't it be under the "Run" menu?

aivarannamaa, 2019-04-04:
I put it under View because it adds or removes a frame in the main window, just like other items in the View menu.

It could be also placed to both menus ...

juanfal, 2019-04-05:
Thanks Aivar, that's a point of view, but IMHO it's a low level, technician's side perspective, more than a user side (normal user flow of actions,)  perspective.
Thank you again for your amazing application I don't get tired of praising.