thonny/thonny #721: fileformat inconsistency between platforms

badukaire, 2019-04-05:
Editing a file with thonny, it looks like it saves files in the current platform format. I have created a file on thonny for windows, git-pushed to github, and then git-cloned in linux (raspbian).

Then did a minor edit with thonny for linux, and git diff reports that the whole file has been changed because of the fileformat linebreaks.

I have checked on the preferences and there is nothing related to fileformat treatment.

I have also checked and there seems to be a somehow related issue #375 but there is not much detail there.

Let me know if this is okay, and if not if I can contribute.

PS: Thanks for your job. I am using thonny to introduce 2+ U14 girls into programming.

aivarannamaa, 2019-04-05:
I don't remember anymore, why did I create #375.

At the moment I don't see why shouldn't it keep original line endings. It would require more work, though, as `tokenize.open`, which is currently used for loading the files, seems to use universal newlines (ie. returns all newlines as `\n`, regardless of the platform). 

It makes sense to keep newlines as `\n` in memory, because Tk on Unix platform displays extra empty lines when `\r\n` is used.

Probably Thonny should consult the target file on save to decide whether to translate in-memory `\n`-s to `\r\n`-s or not.

If you want to implement this then you're welcome!

Ivoz, 2019-04-05:
> Editing a file with thonny, it looks like it saves files in the current platform format. I have created a file on thonny for windows, git-pushed to github, and then git-cloned in linux (raspbian).

Then did a minor edit with thonny for linux, and git diff reports that the whole file has been changed because of the fileformat linebreaks.

Can you specify what format the linebreaks were at each step?
1. when you saved file on windows
2. When looked at raw from inside github (after git has pushed it up to github)
3. The local file after pulling down from git on linux
4. After re-saving with Thonny

Note also that git has a lot of configurability for newlines, you can tell it to munge everything to `\n` or to transform all newlines to what they are "native" on a given platform, IIRC

badukaire, 2019-04-05:
> Can you specify what format the linebreaks were at each step?
>
> 1. when you saved file on windows

On windows, thonny saved the file in DOS fileformat, the girls typed it there.

> 2. When looked at raw from inside github (after git has pushed it up to github)

Not sure how to check this on github website, tried with raw view ...

> 3. The local file after pulling down from git on linux

Still DOS fileformat, gvim opens it as such. I have tried again today.

> 4. After re-saving with Thonny

linux fileformat :(

> Note also that git has a lot of configurability for newlines, you can tell it to munge everything to `\n` or to transform all newlines to what they are "native" on a given platform, IIRC; which depending on your settings can also be changing their form on every push / pull.

I've had to edit the file by hand with gvim, and then manually type `:set fileformat=dos` before committing and pushing back. I am quite sure git has nothing to do here.

I am aware the git for windows / tortoise allows to change the fileformat settings, but it's not the case here.

badukaire, 2019-04-05:
> At the moment I don't see why shouldn't it keep original line endings.

Neither do I :)

It was frustrating for the girls to see that modifying 1 line yielded the whole file modified and watching me open a _weird_ editor. Added more confusion.

> It would require more work, though, as `tokenize.open`, which is currently used for loading the files, seems to use universal newlines (ie. returns all newlines as `\n`, regardless of the platform). Actually, it makes sense to keep newlines as `\n` in memory, because Tk on Unix platform displays extra empty lines when `\r\n` is used.
> 
> Probably Thonny should consult the target file on save to decide whether to translate in-memory `\n`-s to `\r\n`-s or not.

I have done this when handling files, I check if the first lines have 2 or 1 characters for line endings (`\r\n`) then generate output in the same format. But I am sure there must be a better built-in solution.

> If you want to implement this then you're welcome!

I will be glad to do it if you provide me pointers to where files are open and saved.

badukaire, 2019-04-05:
In https://docs.python.org/3/library/functions.html#open it is described how to handle multiplatform handling of opening/writing of file. There is a parameter called _newline_ that sets it. Not sure if this can be applied to thonny.

aivarannamaa, 2019-04-05:
I guess for opening it is best to keep `tokenize.open` because it takes [PEP 263 encoding comments](https://www.python.org/dev/peps/pep-0263/) into account when detecting source encoding. And, as I said, it is good that linebreaks are represent with `\n` in the memory.

Saving takes place in [thonny.code.Editor.save_file](https://github.com/thonny/thonny/blob/master/thonny/code.py#L177), but the conversion from chars to bytes currently occurs in [thonny.codeview.CodeView.get_content_as_bytes](https://github.com/thonny/thonny/blob/master/thonny/codeview.py#L258). I think the `\n` => `\r\n` replacement decision should move to save_file, or you need to add a parameter to get_content_as_bytes.

Ivoz, 2019-04-06:
Note that one solution (although I am not at all against adding a Thonny configuration to change behaviour) is to have git commit all files as LF-only when on windows. And when it checks out files they will be auto-converted to CRLF.

By enabling this git behaviour, then files will be CRLF when retrieved with git on Windows, and LF when retrieved by git on Linux. Therefore the native behaviours (and current existing behaviour of Thonny) or editors should work with no troubles no matter the OS.

It's just setting a global configuration on your Windows install of git -

    C:\> git config --global core.autocrlf true

or you can see a picture of the installer option when installing [here](http://wiki.opf-labs.org/display/SP/Configuring+how+line-endings+are+handled+by+git).

aivarannamaa, 2019-04-06:
@badukaire, I forgot that the user can "Save as" a file, so probably it's better to decide line endings at [load time](https://github.com/thonny/thonny/blob/master/thonny/code.py#L156) after all. Just make sure the encoding is correct and linebreaks stay `\n` in the memory.

And sorry, if it feels like micromanagement :)

badukaire, 2019-04-06:
> Note that one solution/workaround (although I am not at all against adding a Thonny configuration to change behaviour) is to have git commit all files as LF-only when on windows. And when it checks out files they will be auto-converted to CRLF.

Hi Ivoz. Your suggestion is a cool workaround in case of an emergency. But not the proper approach since it is forcing fileformat changes and therefore changing the whole file. Then files edited with python would not be viewed properly with windows non-developer editors.

badukaire, 2019-04-06:
> @badukaire, I forgot that the user can "Save as" a file, so probably it's better to decide line endings at [load time](https://github.com/thonny/thonny/blob/master/thonny/code.py#L156) after all. Just make sure the encoding is correct and linebreaks stay `\n` in the memory.
> 
> Forgive me, if it feels like micromanagement :)

Absolute no micromanagement at all. Any information, direction and suggestion will be more than welcome! Will take a look.

badukaire, 2019-04-06:
I have done some tests outside thonny and I think I have found what is happening, and probably I can come up with a solution. I need to test it inside thonny itself.

badukaire, 2019-04-07:
I have submitted a pull request. Let me know if you have comments.
The point here is how python3 handles file format, it is different (and way better!) to python2 which is what I was used to.
When reading a file, all lines just have a '\n' endline regardless of the fileformat. But you can know the original file format with the _newlines_ attribute.
This attribute could be passed on when saving with the _newline_ parameter, but if saving in binary format like thonny does this is not possible. Therefore it should be converted along with the encoding.

aivarannamaa, 2019-04-07:
@badukaire, I'm happy with #725, thank you!

closed 2019-04-07 by aivarannamaa