thonny/thonny #722: Add an option for specifying how many most recent files to reopen when starting new session

aivarannamaa, 2019-04-05:
By default 5?

badukaire, 2019-04-09:
I would not make this configurable.

I'd go with a fixed (even) number, 4 or 6 look fine.

aivarannamaa, 2019-04-09:
Why even?

lurch, 2020-05-23:
Why have an arbitrary limit? SublimeText re-opens _all_ the files I previously had open when I relaunch it, which is nice.

aivarannamaa, 2020-05-23:
The limit would implement simple and automatic "garbage collection" for less tidy users.

lurch, 2020-05-24:
Ahh, I just found (and enabled) the "Reopen all files from previous session" option! :grinning:  Haha, didn't realise Thonny could already do that.

IMHO if you do add an option "for specifying how many most recent files to reopen when starting new session" it would be nice if there was still an option for "everything" :wink:

Hermann12, 2020-06-02:
Thanks @lurch I searched the same functionality and it's still available. My idea was first make something like " Open Project" where a folder with different files could be open by one click.

aivarannamaa, 2020-06-02:
@Hermann12 , do you know about Thonny's file browser (View => Files)? You can go into project directory and open scripts from there as needed with a double click.

Hermann12, 2020-06-11:
Thanks @lurch I searched the same functionality and it's still available. My idea was first make something like " Open Project" where a folder with different files could be open by one click.

@aivarannamaa ; GREAT, Thanks! I missed this under files. I have got no idea, to search under views.