thonny/thonny #724: Problem when handling DebbugerResponse

Marcel176, 2019-04-06:
Traceback (most recent call last):
  File "/home/marcel/apps/thonny/lib/python3.7/site-packages/thonny/workbench.py", line 1449, in event_generate
    handler(event)
  File "/home/marcel/apps/thonny/lib/python3.7/site-packages/thonny/plugins/debugger.py", line 1055, in _handle_debugger_progress
    _current_debugger.handle_debugger_progress(msg)
  File "/home/marcel/apps/thonny/lib/python3.7/site-packages/thonny/plugins/debugger.py", line 240, in handle_debugger_progress
    self._main_frame_visualizer.update_this_and_next_frames(msg)
  File "/home/marcel/apps/thonny/lib/python3.7/site-packages/thonny/plugins/debugger.py", line 354, in update_this_and_next_frames
    self._next_frame_visualizer.update_this_and_next_frames(msg)
  File "/home/marcel/apps/thonny/lib/python3.7/site-packages/thonny/plugins/debugger.py", line 338, in update_this_and_next_frames
    self._update_this_frame(msg, frame_info)
  File "/home/marcel/apps/thonny/lib/python3.7/site-packages/thonny/plugins/debugger.py", line 862, in _update_this_frame
    DialogVisualizer._update_this_frame(self, msg, frame_info)
  File "/home/marcel/apps/thonny/lib/python3.7/site-packages/thonny/plugins/debugger.py", line 818, in _update_this_frame
    FrameVisualizer._update_this_frame(self, msg, frame_info)
  File "/home/marcel/apps/thonny/lib/python3.7/site-packages/thonny/plugins/debugger.py", line 389, in _update_this_frame
    self._tag_range(frame_info.current_statement, stmt_tag)
  File "/home/marcel/apps/thonny/lib/python3.7/site-packages/thonny/plugins/debugger.py", line 424, in _tag_range
    "%d.0" % self._translate_lineno(text_range.lineno),
AttributeError: 'NoneType' object has no attribute 'lineno'

aivarannamaa, 2019-04-06:
Can you post the program that caused this error?

S11m, 2019-08-03:
Similar thing happened to me while solving https://www.codewars.com/kata/552c028c030765286c00007d/: 

> Traceback (most recent call last):
  File "C:\Users\Aurelius\AppData\Local\Programs\Thonny\lib\site-packages\thonny\workbench.py", line 1449, in event_generate
    handler(event)
  File "C:\Users\Aurelius\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\debugger.py", line 1055, in _handle_debugger_progress
    _current_debugger.handle_debugger_progress(msg)
  File "C:\Users\Aurelius\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\debugger.py", line 240, in handle_debugger_progress
    self._main_frame_visualizer.update_this_and_next_frames(msg)
  File "C:\Users\Aurelius\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\debugger.py", line 354, in update_this_and_next_frames
    self._next_frame_visualizer.update_this_and_next_frames(msg)
  File "C:\Users\Aurelius\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\debugger.py", line 338, in update_this_and_next_frames
    self._update_this_frame(msg, frame_info)
  File "C:\Users\Aurelius\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\debugger.py", line 862, in _update_this_frame
    DialogVisualizer._update_this_frame(self, msg, frame_info)
  File "C:\Users\Aurelius\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\debugger.py", line 818, in _update_this_frame
    FrameVisualizer._update_this_frame(self, msg, frame_info)
  File "C:\Users\Aurelius\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\debugger.py", line 389, in _update_this_frame
    self._tag_range(frame_info.current_statement, stmt_tag)
  File "C:\Users\Aurelius\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\debugger.py", line 424, in _tag_range
    "%d.0" % self._translate_lineno(text_range.lineno),
AttributeError: 'NoneType' object has no attribute 'lineno'

Code:
`
numbers = "2 4 7 8 10"
list_gen = [int(i) for i in [int(s) for s in numbers.split(' ')]]
print (list_gen)
even = [i for i in list_gen if i%2 !=0]
uneven = [i for i in list_gen if i%2==0]
if len(even)==1:
    print (list_gen.index(even[0])+1)
else:
    print (list_gen.index(uneven[0])+1)
`

aivarannamaa, 2019-08-03:
@S11m, Thank you! I could fix your case.

closed 2019-08-08 by aivarannamaa