thonny/thonny #729: default/new file format option

badukaire, 2019-04-08:
This issue is absolutely not urgent, and perhaps not even necessary.

It may be useful to allow a configuration option for forcing the default fileformat? Perhaps some editors force it (thonny did it in windows until a few days ago).

I have checked the behaviour on notepad++ and gvim. Both editors respect the file format on already created files.

In notepad++ it is possible to set the format for new files (see attached image), with 3 possible values: DOS/unix/MAC(!).

In gvim, adding a line like `set fileformat=unix` in the configuration file makes a similar thing unless this variable is overridden by the user or even by opening a file.

[image]

So, in case you find this option useful, would you implement it as in notepad++ or do you have other ideas?

aivarannamaa, 2019-04-09:
It may be useful for some people. I'm using such option myself in my Windows Eclipse, but I'm not Thonny's target group :) Anyways, I'm not against it.

Looks like "Editor" config page has room for one more combobox for specifying linebreaks for new files. What do you think of choices "Platform default", "Windows (CR LF)" and "Unix (LF)"?

You may want to use this method for creating and placing the combobox: https://github.com/thonny/thonny/blob/master/thonny/config_ui.py#L119

juanfal, 2019-04-09:
Macintosh CR line ending dates back to Mac OS Classic, 20 years ago left. Since then, only for compatibility, but no editor on Mac uses it anymore. There was another, then amazing, operating system that used to use CR line endings, VAX VMS (1977–2000)

badukaire, 2019-04-09:
> It may be useful for some people. I'm using such option myself in my Windows Eclipse, but I'm not Thonny's target group :) Anyways, I'm not against it.
>
> Looks like "Editor" config page has room for [...]

Yes, this approach looks better than the one that notepad++ uses. Needless to say the _platform default_ option would be the active one by default. Very important to specify _new files_ so that the file format is never modified.

Now that you talk about thonny's _editor config page_ I (as a new, fresh user) find that its options tabs are not very intuitive regarding the options contained into them. I can open another issue for not mixing subjects here.

> You may want to use this method for creating and placing the combobox: https://github.com/thonny/thonny/blob/master/thonny/config_ui.py#L119

Thanks, I will take a look. I may come up with more questions, like how are the settings shared between all the classes, etc.

Shall a specific, new development branch be created or working in master is preferred?

aivarannamaa, 2019-04-09:
> Now that you talk about thonny's editor config page I (as a new, fresh user) find that its options tabs are not very intuitive regarding the options contained into them. I can open another issue for not mixing subjects here.

Please do!

> Shall a specific, new development branch be created or working in master is preferred?

So far I have used only master for creating new features and this seems to work well enough.