thonny/thonny #730: options dialog refactoring

badukaire, 2019-04-09:
# options dialog refactoring

As I have mentioned in some other issue, being myself a new, fresh user, I find that the settings tabs is not very intuitive regarding what to find on each tab.

To start with, the tabs seem to be displayed in alphabetical order. Is this intentional? I believe that tabs with the most used options should be at the beginning (left), whereas the most rare / specific should be at the end (right).

I also find that the options contained in some tabs are not really matching their tab name and perhaps could be moved to another tab.

Follows a proposal for refactoring the options dialog. Please bear in mind this is only a proposal, that it is something subjective and personal. Of course it is not required that we all agree on everything I suggest. :)

## tab names

I would make the following changes to the tab names:

* editor => syntax : most options there are related to syntax & completion
* +editor : create an _editor_ tab for actual editor options
* theme & font => appearance & theme : later I suggest to move some options there
* general => ? : not sure if a better descriptive name can be found, perhaps _others_?

## tab order

I would place the tabs in the following order (left to right):

* appearance & themes
* syntax
* editor
* assistant
* general
* debugger
* interpreter
* terminal

I am not really sure about the last 4, I haven't used much these features yet.

## moved options

I would move the following options:

* _show line numbers_ in **current** _editor_ tab would be moved to the **new** _editor_ tab, along with the (new?) file format options
* _UI-related_ options in _general_ would be moved to _appearance and themes_ tab
* perhaps the _interpreter_ tab options can be moved to _general_, but the interpreter is a _very important_ option that must be found quickly in case a user needs it

I agree on what it is said in #720, I don't find it intuitive to have the command line arguments in the _view_ menu. In some IDEs like Microsoft's visual studio they are also a program option.

aivarannamaa, 2019-04-17:
Thank you for the proposal. I don't feel too bad about current situation, but I'll think about changing something in 2.2.

The tabs are ordered alphabetically, because the tabs are added by plug-ins and there is no central piece of code which knows all tabs. But it may be good idea to change config page registration such that plug-ins can provide a preferred rank for the tab.

aivarannamaa, 2024-05-30:
I think I'll leave the dialog as is for now.

closed 2024-05-30 by aivarannamaa