thonny/thonny #731: [GUI Issue] [Included Images]: Right click menu disappears immediately and executes unwanted Undo action

vnktsh, 2019-04-10:
Hello,

Thonny right-click menu in editor completely breaks when using any theme other than "Enhanced Calm". The right click menu disappears immediately and unwanted action "undo" is executed.

The following is with theme "Clean Dark". But the issue can be reproduced with any other theme other than default.

UI Theme: Clean Dark  |  Syntax Theme: Default Dark
[image]

**No Issue with default theme:**
UI Theme: Enhanced Calm  |  Syntax Theme: Default Light
[image]

### Steps to reproduce:
1.  Open editor with any file
2.  Goto Menu bar > Tools > Options ; In options windows go to Themes & Font
3. Select "UI Theme: Clean Dark  |  Syntax Theme: Default Dark"
4. Restart thonny
5. Right click anywhere on editor. The pop-up menu is broken and disappears and executes "undo"

### Details:

- Thonny: 3.1.2
- Python: 3.5.2
- Tk 8.6.5
- Linux (64 bit)
- OS: Ubuntu 16.04

aivarannamaa, 2019-04-10:
Thank you for the report! I'll look into this today

vnktsh, 2019-04-12:
Thank you for the attention, and thank you in general for developing this awesome IDE.

commit be0e2504 (2019-04-14): FIXES #731: Right click menu disappears immediately
closed 2019-04-14 by aivarannamaa