thonny/thonny #735: Introduce a code formatting policy for Thonny's source code

aivarannamaa, 2019-04-17:
I don't like strict rules about formatting, but it would be nice if PR diffs did not contain noise related to whitespace.

I suppose everybody agrees about trimming trailing spaces from non-empty lines.

Is it common to also trim all spaces/tabs from "empty" lines? Even if the empty lines are in the middle of an indented block?

cspaier, 2019-04-17:
I have not much experience here and tend to follow PEP8 and FLake8:

From [PEP8](https://www.python.org/dev/peps/pep-0008/#other-recommendations), one can read:

> Avoid trailing whitespace anywhere. Because it's usually invisible, it can be confusing: e.g. a backslash followed by a space and a newline does not count as a line continuation marker. Some editors don't preserve it and many projects (like CPython itself) have pre-commit hooks that reject it.

[Flake8](https://lintlyci.github.io/Flake8Rules/rules/W293.html) clearly reject blank line with white space.

aivarannamaa, 2019-04-18:
Alright, in this case I need to make a commit which removes all trailing whitespace from the codebase.

Ilis, 2019-04-21:
PEP8 says comments should start with `"# "` (space after hash), and I suggest Add comment and Toggle comment follow this rule and format comments accordingly.

aivarannamaa, 2019-04-21:
> PEP8 says comments should start with "# "

My idea was to distinguish documentation comments from comments meant for disabling code. But it looks like other IDE-s include the space, so I don't mind changing it.

Ivoz, 2019-04-22:
You could setup travis CI to run a lint over PRs with a pep8/flake8/etc of your liking if you wanted.

aivarannamaa, 2019-05-29:
I chose Black with 100 char line length -- more details here: https://github.com/thonny/thonny/blob/master/CONTRIBUTING.rst#code-format

Pull requests will be now checked with this formatter as well.

closed 2019-05-29 by aivarannamaa