thonny/thonny #739: Deploying Thonny with packages preinstalled

AndrewBreyen, 2019-04-26:
Hello,

I am working on packaging Thonny [with a bunch of packages preinstalled](https://gist.github.com/AndrewBreyen/3041de3fb1372fc9768c1accd8103fcb#file-thonnypackages19-20-txt). 

I am looking at [the Deployment Options page](https://github.com/thonny/thonny/wiki/DeploymentOptions) but I am very confused.

I believe that I need to change the default user directory, and with my understanding the best way to do that is with customize.py. I am struggling to find the customize.py file on a system that I manually installed the latest version of Thonny on. 

Customize.py is not located where [it says it should be](https://github.com/thonny/thonny/wiki/DeploymentOptions#customizepy): `C:\Users\**CURRENTUSER**\AppData\Local\Programs\Thonny\Lib\site-packages\thonny`. I don't even have a `programs` folder in `C:\Users\**CURRENTUSER**\AppData\Local`.

How can python be deployed with alriris?

aivarannamaa, 2019-04-28:
Hi!
The file does not exist in standard installation, you need to create it yourself. Take a look at https://github.com/thonny/thonny/releases/download/v3.1.2/thonny-3.1.2-portable-win.zip

aivarannamaa, 2019-04-28:
Unfortunately I don't know about alriris (altiris?)

closed 2019-06-20 by aivarannamaa