thonny/thonny #740: Error when creating directory

MsAnanke, 2019-04-28:
When trying to create a directory with the following code I get an error message (this same code works with Anaconda):

**import os

chemin='G:\\Courses\\'
dat_course='22-avril-2019'

dirName=chemin+dat_course+'CTURF'

try:
        os.makedirs(dirName)    
        print("Directory " , dirName ,  " Created ")
    except FileExistsError:
        print("Directory " , dirName ,  " already exists")** 

I get the following error message (and a window opens to try to save my .py file):

Traceback (most recent call last):
  File "C:\Users\Utilisateur\AppData\Local\Programs\Thonny\lib\tkinter\__init__.py", line 1705, in __call__
    return self.func(*args)
  File "C:\Users\Utilisateur\AppData\Local\Programs\Thonny\lib\site-packages\thonny\running.py", line 317, in _cmd_run_current_script
    self.execute_current("Run")
  File "C:\Users\Utilisateur\AppData\Local\Programs\Thonny\lib\site-packages\thonny\running.py", line 282, in execute_current
    script_dir = normpath_with_actual_case(os.path.dirname(filename))
  File "C:\Users\Utilisateur\AppData\Local\Programs\Thonny\lib\ntpath.py", line 221, in dirname
    return split(p)[0]
  File "C:\Users\Utilisateur\AppData\Local\Programs\Thonny\lib\ntpath.py", line 183, in split
    p = os.fspath(p)
TypeError: expected str, bytes or os.PathLike object, not NoneType

aivarannamaa, 2019-04-28:
What is the name of your script file? Is it located on a network path (`\\servername\path`)? Which Thonny version are you using?

MsAnanke, 2019-04-28:
Hello
Script file name is: Pronostics_www_canalturf_com.py
It is not located on a network path
Version of Thonny is: 3.1.2 on Windows 10 64 bits, Python 3.7.3 32 bits, Tk 8.6;8

aivarannamaa, 2019-04-28:
Does restarting Thonny help?

aivarannamaa, 2019-04-28:
For your information -- the error message seems to happen even before Thonny executes your code.

MsAnanke, 2019-04-28:
Restarting Thonny does not help.
But I think I found what is creating the problem (but I don't know why): my script was not saved yet when I tried to run it and got the error. 
If I save the script before trying to run it it works. No clue why.

aivarannamaa, 2019-05-03:
Now I could finally reproduce it. I assume you tried to run an unnamed script and when Thonny asked to save it you chose "Cancel".

Thonny needs the script to be saved before running it, but after cancelling the Save dialog it should have  done nothing instead of showing the error.

aivarannamaa, 2019-06-07:
Duplicate of #692

closed 2019-06-07 by aivarannamaa