thonny/thonny #743: Update packages using the internal Manage packages for ...

Cornelliuss, 2019-04-28:
Okay, I am very new to this and would like to know how or if the mange packages work in Thonny.  I tried to upgrade parso, jedi, and pylint.  I managed to download the "packages" from the internet but not sure how to import them into the Thonny environment.  Any help will be much appreciated.  

I tried using the pip command from the terminal but can't get that to work either.

aivarannamaa, 2019-05-03:
Did you try "Tools => Manage packages" ?

Cornelliuss, 2019-05-07:
Yes I tried the manage packages under Tools, but the version number stays the same even after restarting the program.

Regards,

Cornelius du Preez

Assistant-Professor
Dongyang University, Yeoungju, South Korea
Mobile: +82 (0)10-2660-3605

aivarannamaa, 2019-05-07:
>  I managed to download the "packages" from the internet 

How did you do that? Did you see the pip output dialog after pressing "Install" or "Update"? What did it say?

Cornelliuss, 2019-05-07:
Good day,

I’ll use an example with some pictures:

I want to install ‘jedi’

1. I go to tools -> Manage packages
2. type jedi and hit enter or find package from PYPI

There is no install option.

I read it requires ‘parso’ so I typed it in the find package from PyPi and click on the button

There I saw an upgrade button so I clicked on it and it installed parso,
but I can still not install ‘jedi’.

Not sure if this will help I run Thonny 3.1.0 on a RaspBerryPi 3B+.

Regards,

Cornelius du Preez

Assistant-Professor
Dongyang University, Yeoungju, South Korea
Mobile: +82 (0)10-2660-3605

aivarannamaa, 2019-05-07:
`jedi` should already be installed, because Thonny uses it. It should be available when you are using default interpreter ("Same as front-end"). Did you try using it in your programs?

Cornelliuss, 2019-05-07:
How does one upgrade it to the newer version?
Why does the install button disappear?

As I type in the front end, the auto complete does not work properly. I'm
running it on a raspberry pi 3B+.  How do one upgrade from python 3.5 to
python 3.7 on raspberry?

Thanks die your support.

-- 

Cornelius du Preez

Punggi-Eup, South Korea
Mobile: +82 (0)10-2660-3605
Skype: Cornelliuss

------------------------------------------------------------
-----------------------------------------------
Always be joyful.  Praying without giving up.  In every circumstance give
thanks:
for this is the will of God in Christ Jesus concerning you.
1 TH 5:16-18
-----------------------------------------------------------------------------------------------------------

Cornelliuss, 2019-05-08:
Mmm... As I responded from my email account it added lots of not needed information too, so I removed the responses.

Yes, I used the "Tools -> Manage packages" Some times there is an install button and then it installs the packages, but the install button disappears, so one can't upgrade to newer package versions. 

I was following the setup for MS Visual Studio where they explained additional packages to be installed.  - parso, jedi, and... However, as I am using the Raspberry PI 3B+ MS VS is a total overkill and I like Thonny's layout.  So I am trying to load get Thonny to work like some of what I noticed in VS.  Maybe my understanding of how 'parso' and 'jedi' work is wrong.  Is it supposed to display the possible options file._________ and then give a list of possible options?

As I type in the front end, the autocomplete does not work properly. 
How do one upgrade from python 3.5.3 to python 3.7.2 on the raspberry?

Thanks, for your support.

aivarannamaa, 2019-05-08:
> Maybe my understanding of how 'parso' and 'jedi' work is wrong. Is it supposed to display the possible options file._________ and then give a list of possible options?

`jedi` is just a library for collecting the options to show, the presentation must be implemented by the IDE. In Thonny you need to press Ctrl+Space in order to invoke code completion.

> How do one upgrade from python 3.5.3 to python 3.7.2 on the raspberry?

This is not easy. Most likely you need to build it from source. In this case you probably should not try to replace the system python, but add the new one as alternative.

When Raspberry moves from Debian Stretch to Debian Buster, you'll get Python 3.7 officially.

Cornelliuss, 2019-05-10:
Thanks for your assistance Aivar Annamaa.  I'll wait for the release of Debian Buster.

closed 2019-05-10 by Cornelliuss