thonny/thonny #746: Error on Win 8.1

Remuar, 2019-04-28:
Hola,
I got following error with Thonny 3.12 & 3,08 on Win 8.1
as admin or normal user with your last install.
.... any hint???

>>>
Traceback (most recent call last):
  File "C:\Program Files\Thonny\lib\site-packages\thonny\workbench.py", line 465, in _start_runner
    self._runner.start()
  File "C:\Program Files\Thonny\lib\site-packages\thonny\running.py", line 79, in start
    self.restart_backend(False, True)
  File "C:\Program Files\Thonny\lib\site-packages\thonny\running.py", line 431, in restart_backend
    self._proxy = backend_class(clean)
  File "C:\Program Files\Thonny\lib\site-packages\thonny\running.py", line 995, in __init__
    CPythonProxy.__init__(self, get_frontend_python())
  File "C:\Program Files\Thonny\lib\site-packages\thonny\running.py", line 592, in __init__
    self._start_new_process()
  File "C:\Program Files\Thonny\lib\site-packages\thonny\running.py", line 759, in _start_new_process
    debug("Starting the backend: %s %s", cmd_line, get_workbench().get_cwd())
  File "C:\Program Files\Thonny\lib\site-packages\thonny\workbench.py", line 1235, in get_cwd
    return normpath_with_actual_case(os.path.expanduser("~"))
  File "C:\Program Files\Thonny\lib\site-packages\thonny\common.py", line 208, in normpath_with_actual_case
    assert os.path.exists(name)
AssertionError

aivarannamaa, 2019-04-28:
This stacktrace indicates a very strange condition, as if your home directory did not exist.

Do you happen to have another Python interpreter installed in this machine? If yes, then what does following code print:

```
import os.path
print(os.path.expanduser("~"))
```
Does the directory, that gets printed, exist?

aivarannamaa, 2019-04-28:
Alternatively, where do you get when you enter `%USERPROFILE%` into Windows Explorer status bar?

[image]

Remuar, 2019-05-01:
I test with IDLE and got following message before IDLE starts
>>>

 Warning: os.path.expanduser("~") points to
 C:\SPB_Data,
 but the path does not exist.

<<<<

So you´r rigth path disapeared.
I will try to re create and let you know soon if it works.
Thank´s for your kind answer

Remuar, 2019-05-01:
Problem solved.... I redirected the home

[cid:60dbbf54-72ba-4d66-a7c8-149bfd0a77ff]

________________________________

aivarannamaa, 2019-05-01:
Great! I added similar check and warning to Thonny.

commit ce0eced8 (2019-05-01): FIXED #746: Warn if user home is missing
closed 2019-05-01 by aivarannamaa