thonny/thonny #750: Thonny test failures with Python 3.8

hroncok, 2019-05-03:
This is Fedora 31, Python 3.8.0a3, Thonny 3.1.2:

```
+ xvfb-run py.test-3 --pyargs thonny
============================= test session starts ==============================
platform linux -- Python 3.8.0a3, pytest-4.4.1, py-1.8.0, pluggy-0.9.0
rootdir: /builddir/build/BUILD/thonny-3.1.2
collected 26 items

thonny/plugins/micropython/api_stubs/lcd160cr_test.py ...                [ 11%]
thonny/test/test_ast_utils.py F                                          [ 15%]
thonny/test/test_ast_utils_mark_text_ranges.py FFFFFFFFFFFFFFFFF         [ 80%]
thonny/test/test_common.py .                                             [ 84%]
thonny/test/plugins/test_coloring.py .                                   [ 88%]
thonny/test/plugins/test_locals_marker.py .                              [ 92%]
thonny/test/plugins/test_name_highlighter.py .                           [ 96%]
thonny/test/plugins/test_paren_matcher.py .                              [100%]

=================================== FAILURES ===================================
_______________________ test_pretty_without_end_markers ________________________

    def test_pretty_without_end_markers():
        p = pretty(
            ast.parse(
                dedent(
                    """
        age = int(input("Enter age: "))
        if age > 18:
            print("Hi")
        else:
            print("Hello!", end='')
            print("What's your name?")
        """
                ).strip()
            )
        )
    
>       assert (
            p
            == """/=Module
        body=[...]
            0=Assign @ 1.0
                targets=[...]
                    0=Name @ 1.0
                        id='age'
                        ctx=Store
                value=Call @ 1.6
                    func=Name @ 1.6
                        id='int'
                        ctx=Load
                    args=[...]
                        0=Call @ 1.10
                            func=Name @ 1.10
                                id='input'
                                ctx=Load
                            args=[...]
                                0=Str @ 1.16
                                    s='Enter age: '
                            keywords=[]
                    keywords=[]
            1=If @ 2.0
                test=Compare @ 2.3
                    left=Name @ 2.3
                        id='age'
                        ctx=Load
                    ops=[...]
                        0=Gt
                    comparators=[...]
                        0=Num @ 2.9
                            n=18
[... 1127 lines cut ...]
>       assert actual_pretty_ast.strip() == expected_pretty_ast.strip()
E       AssertionError: assert '/=Module\n  ...pe_ignores=[]' == '/=Module\n   ...@ 5.0  -  5.4'
E         Skipping 215 identical leading characters in diff, use -v to show
E         -     value=Constant @ 2.7  -  4.8
E         ?           ^^^^ ^^^
E         +     value=Str @ 2.7  -  4.8
E         ?           ^ ^
E         -                 value='first\nsecond\nthird'
E         ?                 ^^^^^...
E         
E         ...Full output truncated (9 lines hidden), use '-vv' to show

thonny/test/test_ast_utils_mark_text_ranges.py:516: AssertionError
=============================== warnings summary ===============================
thonny/roughparse.py:8
  /builddir/build/BUILD/thonny-3.1.2/thonny/roughparse.py:8: DeprecationWarning: Using or importing the ABCs from 'collections' instead of from 'collections.abc' is deprecated, and in 3.8 it will stop working
    from collections import Mapping

thonny/test/test_ast_utils_mark_text_ranges.py::test_single_assignment
  /builddir/build/BUILD/thonny-3.1.2/thonny/ast_utils.py:181: DeprecationWarning: The 'warn' method is deprecated, use 'warning' instead
    logger.warn("asttokens not found, using deprecated alternative")

-- Docs: https://docs.pytest.org/en/latest/warnings.html
=============== 18 failed, 8 passed, 2 warnings in 0.93 seconds ================
```

hroncok, 2019-05-03:
Partially, this is caused by: https://github.com/python/cpython/pull/9445

> Class `ast.Constant` is now used for all constants. Old classes `ast.Num`, `ast.Str`, `ast.Bytes`, `ast.NameConstant` and `ast.Ellipsis` are still available, but they will be removed in future Python releases.

aivarannamaa, 2019-05-03:
Thank you for bringing my attention to it!

I'll start working on 3.8 compatibility when 3.8b1 comes out.

hroncok, 2019-06-07:
3.8.0.b1 is out. I cannot attempt to rebuild thonny, as parso fails to build, so I cannot post any new logs yet.

aivarannamaa, 2019-06-07:
Most likely I need to update something in Thonny, but I'll write down a related issue here.

Since 3.1 Thonny relies on https://github.com/gristlabs/asttokens for marking source range information in AST nodes. Hopefully they will include Python 3.8 compatibility soon (https://github.com/gristlabs/asttokens/pull/29). 

It looks like AST nodes in Python 3.8 already have end_lineno and end_col_offset attributes but I'm not sure if they are reliable (lineno and col_offset in earlier versions did require some tweaking). I'd like to avoid worrying about this and delegate this problem to asttokens.

Can you upgrade Fedora's python3-asttokens if the upstream releases new version?

hroncok, 2019-06-07:
I can.

closed 2019-06-14 by aivarannamaa