thonny/thonny #751: Breaking a line with \ causes green syntax error after a " because it doesn't see the the first "

bjelkea, 2019-05-05:
Running this will cause the green syntax error in the editor when everything is ok:

print(f"this causes green \
syntax error")

aivarannamaa, 2019-05-05:
Thank you for the report!

I didn't even know that one can break a string literal like this.

raenye, 2020-10-18:
Here it worked alright -- green syntax error disappeared the moment the string literal was terminated.

But the bug I can produce is when the string literal remains unterminated (see screenshot below).
The green background disappears after a keypress, though.

```python
"unterminated string
```
[image]