thonny/thonny #755: Multiple lab iMacs with Thonny, package installing

juanfal, 2019-05-07:


aivarannamaa, 2019-05-07:
I suppose it is possible if you know the location of the bundled Python interpreter (`import sys; print(sys.executable)`). 

You could try something like this:

```
/path/to/python -m pip install package1 package2
```

This way the packages would end up in `site-packages` inside the application bundle, not under user home (as would happen if user installed them via Tools => Manage packages).

juanfal, 2019-05-07:
Thanks for a so quick answer!

So I should spread-launch for every machine through ARD:

    /Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.7/bin/pip3 install matplotlib

Is it right?  I understand the destination folders is bonded to:

    /Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages

Despite that, in that directory doesn't exist the `numpy` for example, library I did install time ago and you can `import numpy` at the Thonny shell.  How is that possible?

aivarannamaa, 2019-05-07:
> Despite that, in that directory doesn't exist the numpy for example, library I did install time ago and you can import numpy at the Thonny shell. How is that possible?

If you installed it via the dialog, then it was installed with `pip install --user`, ie. it should be in `~/Library/Python/X.Y/lib/python/site-packages` (https://docs.python.org/3/library/site.html#site.USER_SITE)

aivarannamaa, 2019-05-07:
You can also check the installation location in Thonny's pip dialog when you click on the name of installed package.

juanfal, 2019-05-07:
Yes!! Thank you

It says that `numpy` is installed in:

    ~/Library/Python/3.7/lib/python/site-packages

so I understand that if I replace Thonny by new versions, I wouldn't need to reinstall these packages. Could I get a list of those installed in that home Library?

aivarannamaa, 2019-05-07:
> I get a list of those installed in that home Library?

You could do `ls ~/Library/Python/3.7/lib/python/site-packages`. It may show 2 directories per package, though. `pip3 list --user` should work as well

> Does it mean you could install in a shared by Thonny with Terminal OSX, using: `pip3 install --user`

That's right

Ivoz, 2019-07-21:
Could thonny store a list a of "interpreters it knows about (at last run)"  (like those added by a custom interpreter entry, and an in-built entry if it can find it, etc, then have a convenience command-line function that would pass a requirements file to a `{that_python} -m pip -r` invocation?

aivarannamaa, 2019-07-21:
@Ivoz, do I understand correctly that you want to manage packages for an interpreter, which is not currently selected as Thonny's interpreter?

Ivoz, 2019-07-22:
Yes, that is part of my suggestion. Allow to 'manage packages' for any normal interpreter [i.e not ephemeral like a MicroPython] that Thonny would let you select in its options.

Although that could also be me trying to cover more use cases than needed. You could also implement this feature only for thonny's (current selected? default?) interpreter and that could perhaps solve 90% of use cases for now. i.e run this command and then next time someone (student) starts up thonny they have access to x,y,z packages in the current interpreter out of the gate.

aivarannamaa, 2019-07-22:
> I agree probably the simple way first is better, implement it just for current interpreter for now

Sorry, I still don't get it -- implement what and which problem would it solve?

Ivoz, 2019-07-23:
Sorry again, I mean a CLI-runnable command something like 

    $ {thonny / python -m thonny} --install-reqs=requirements.txt

 that would pass that requirements file to a pip to install in Thonny's current interpreter. Thereby allowing one to script pre-populating thonny installs with installed packages.

aivarannamaa, 2019-07-23:
Thanks! Now I understand...

I think in this situation one knows which interpreter is being used by Thonny and can do something like this instead (if it's bundled interpreter):

```
$ /thonny/bin/pip install -r requirements.txt
```

Now, the problem is that in Mac and Linux bundles, this currently requires (DY)LD_LIBRARY_PATH to be set against Thonny's lib dir. But I can change [pip.sh](https://github.com/thonny/thonny/blob/master/packaging/pip.sh) (which gets copied to bundle's bin directory) so that it sets this variable automatically.

aivarannamaa, 2019-08-09:
I was wrong about requiring DYLD_LIBRARY_PATH to use bundled pip directly. At least in version 3.2 you can use something like that for installing into bundled site-packages:

`~/Desktop/Thonny.app/Contents/Frameworks/Python.framework/Versions/Current/bin/pip3 install ...`

closed 2019-08-09 by aivarannamaa