thonny/thonny #758: Code as a flowchart

antonio-c-mariani, 2019-05-08:
It would be nice if the python code could be presented as a flowchart. For beginners this feature is very useful as a way to better understand the code, specially for nested structures.

aivarannamaa, 2019-05-09:
Good idea! 

It's probably cheaper to provide the chart as a html document, because it's easier to find suitable graphic libraries for JavaScript rather than Tkinter.

At the same time, Tkinter implementation could be integrated more cleanly.

I found a relevant resource: http://codimension.org/assets/cdm/visualization/Visualization.pdf, but I'm not sure I want to use their approach in Thonny. They only deconstruct the `if`-statement into a regular flowchart fragment.  With all other compound statements they have, IMO, "cheated" by constructing a specific graphic notation, which don't offer big benefits over regular Python syntax.

But general, their IDE is really interesting: http://codimension.org/

aivarannamaa, 2021-01-19:
I'll leave this for Codimension

closed 2021-01-19 by aivarannamaa