thonny/thonny #760: Can't install package (gcc error) on clean install

wencha, 2019-05-20:
I open up Thonny for the first time and tried to install pcapy library from the pypi package manager and it fails saying:
```
    creating build/temp.linux-x86_64-3.7
    gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -fPIC -I/home/franzisca/apps/thonny/include/python3.7m -c pcapdumper.cc -o build/temp.linux-x86_64-3.7/pcapdumper.o
    pcapdumper.cc:10:10: fatal error: Python.h: No such file or directory
     #include <Python.h>
              ^~~~~~~~~~
    compilation terminated.
    error: command 'gcc' failed with exit status 1
```
I'm on Xubuntu 18 and have python3-dev installed. Also failed with psutil. Also tried opening sudo thonny. gcc version 7.4.0. Also tried installing python3.7-dev. Error persists. I've installed thonny with normal user running install script.
Thanks

aivarannamaa, 2019-05-20:
Thonny + Python bundles don't include all files required for building native packages.

If you have suitable Python available in your system, you can pip-install Thonny against this (eg. `/usr/bin/python3 -m pip install thonny`.

wencha, 2019-05-20:
Yes! Great. I didn't see that on the docs. Thx

closed 2019-09-01 by aivarannamaa