thonny/thonny #761: Never seen before bug

Andre50501, 2019-05-21:
Thonny 3.?.?  worked perfectly fine on my MacBook Pro running Mojave until yesterday. When I launched it, it looked like this:
[image]

Cant write any code, menus are all greyed out. I can't even quit... I have to Force quit through the Apple menu. I downloaded version 3.1.2 and re-installed it on my machine with exactly the same issue. I have Python 3.7 installed and IDLE and Spyder work fine.

Does anyone have any ideas?

Andre

juanfal, 2019-05-21:
No problems at all in my Mojave 10.14.6 beta (18G29g), and never had so far, 🤞

aivarannamaa, 2019-05-21:
@Andre50501, can you please check into `~/Library/Thonny` and send me the log-files and configuration.ini?

Andre50501, 2019-05-21:
Hi,

Here are the files you’re requested. Thanks for looking into this.

Andre

aivarannamaa, 2019-05-21:
@Andre50501, did you forget to attach the files?

Andre50501, 2019-05-21:
Hi,

They were in a folder called “Thonny”. Didn’t you see the folder? Here are the files attached individually.

Andre

aivarannamaa, 2019-05-21:
No, I don't see the files neither in GitHub nor in my e-mail client.

Andre50501, 2019-05-22:
Hi, can you see the files now?

Andre

aivarannamaa, 2019-05-22:
No, I still don't see them.

Trying to attach a file myself:
[test.txt](https://github.com/thonny/thonny/files/3207644/test.txt)

Andre50501, 2019-05-23:
Very weird. I got your file and I can see my attached files in my Sent Messages. And I also sent the email and attachments to another email address that belongs to me without any problems. 

Maybe your server doesn’t like .ini or log files?

Hi can I send the files to Github?

Andre

aivarannamaa, 2019-05-23:
> can I send the files to Github?

Sure. If you reply in GitHub, you can drag your files to the comment box.

Andre50501, 2019-05-23:
OK so here was the problem. I couldn't upload configuration file because Github says: "We don't support that file type. Try again with a GIF, JPG ....XLSX or Zip".  So I saved it as a .txt file and its now attached.

The .log files are all empty and they are not attaching.

Regards,

Andre
[configuration.txt](https://github.com/thonny/thonny/files/3211126/configuration.txt)

aivarannamaa, 2019-05-23:
Thanks! Unfortunately I don't see anything suspicious in the configuration file. I was hoping to get more information from the logs, but if you had to force quit Thonny, then it might have been unable to save the logs.

Can you please run Thonny from the terminal? It may give some hints into standard output. See here for instructions: https://github.com/thonny/thonny/wiki/Troubleshooting#open-in-terminal-macos

Is it possible that when you open Thonny, it tries to show an information (or error) dialog, but it's hidden behinde the main window? I've seen Tkinter programs occasionally doing this and it would explain why you could not interact with the window.

aivarannamaa, 2019-09-01:
@Andre50501, Can you please check whether [Thonny 3.2.0](https://github.com/thonny/thonny/releases/download/v3.2.0/thonny-3.2.0.dmg) works for you?

aivarannamaa, 2021-01-14:
Hoping it is fixed by now

closed 2021-01-14 by aivarannamaa