thonny/thonny #767: Cannot install packages from PyPI, because libffi.so.6 not found

cfeck, 2019-05-29:
Using Thonny 3.1.2 on Linux x86_64 (openSUSE Tumbleweed), I cannot install packages via the "Tools/Manage Packages..." installer.

Trying to install "pygame" from PyPI, I get the error depicted below. My system has "libffi.so.7" installed; it is suggested to ship the Linux installers for Thonny with all needed dependencies.

Traceback (most recent call last):
  File "/home/pepo/apps/thonny/lib/python3.7/runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "/home/pepo/apps/thonny/lib/python3.7/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/home/pepo/apps/thonny/lib/python3.7/site-packages/pip/__main__.py", line 16, in <module>
    from pip._internal import main as _main  # isort:skip # noqa
  File "/home/pepo/apps/thonny/lib/python3.7/site-packages/pip/_internal/__init__.py", line 40, in <module>
    from pip._internal.cli.autocompletion import autocomplete
  File "/home/pepo/apps/thonny/lib/python3.7/site-packages/pip/_internal/cli/autocompletion.py", line 8, in <module>
    from pip._internal.cli.main_parser import create_main_parser
  File "/home/pepo/apps/thonny/lib/python3.7/site-packages/pip/_internal/cli/main_parser.py", line 12, in <module>
    from pip._internal.commands import (
  File "/home/pepo/apps/thonny/lib/python3.7/site-packages/pip/_internal/commands/__init__.py", line 6, in <module>
    from pip._internal.commands.completion import CompletionCommand
  File "/home/pepo/apps/thonny/lib/python3.7/site-packages/pip/_internal/commands/completion.py", line 6, in <module>
    from pip._internal.cli.base_command import Command
  File "/home/pepo/apps/thonny/lib/python3.7/site-packages/pip/_internal/cli/base_command.py", line 18, in <module>
    from pip._internal.download import PipSession
  File "/home/pepo/apps/thonny/lib/python3.7/site-packages/pip/_internal/download.py", line 38, in <module>
    from pip._internal.utils.glibc import libc_ver
  File "/home/pepo/apps/thonny/lib/python3.7/site-packages/pip/_internal/utils/glibc.py", line 3, in <module>
    import ctypes
  File "/home/pepo/apps/thonny/lib/python3.7/ctypes/__init__.py", line 7, in <module>
    from _ctypes import Union, Structure, Array
ImportError: libffi.so.6: cannot open shared object file: No such file or directory

aivarannamaa, 2019-05-30:
Thank you for the report!

I could reproduce this in Tumbleweed. When I'm about make the next release, I'll see whether simply bundling libffi.so.6 helps.

aivarannamaa, 2021-05-20:
I finally got it sorted out. The Linux bundles at https://github.com/thonny/thonny/releases/tag/v3.3.10 now include `libffi.so.6.0.4` and its symlink `libffi.so.6`.

closed 2021-05-20 by aivarannamaa